makerbase-mks/MKS-Robin #1: MKS-Robin version 1.0 firmware update with PID enabled

3Dangel, 2017-02-08:
Dear MKS,

I saw that we have updated with V2.01 firmware with the PID to better control the hotend.

would you please upload for the v1.01 's ?
thanks

Robert

makerbase-mks, 2017-02-13:
hi,you can use V2.0.1, it is the newest

3Dangel, 2017-02-13:
Thanks!!!
I received V 1.0.2
From your company. 
My board is the first batch V1.0

Would you be able to upload the v1.0.2?

It will correct the PID temperature control to +-0.8C!!!

Also, there are two extra 12/24 V socket. How could I control them to enable for second fan and LED light?

Thanks a lot!!!

Robert

makerbase-mks, 2017-02-13:
Hello, the V1.0.2 version firmware has been uploaded to Gihub. 
The extra 12V/24V on the controller board was brought from the power supply, so it is uncontrolable.

3Dangel, 2017-02-13:
Great news for other users!!!
Thanks for uploading. 
Others has confirmed that it is working great for the PID!!

Is the WIFI module need to be set with special setting ? I bought the MKS- WIFI module and would like to set it up with Robin board. 
在這裡先祝福你們新年快樂萬事如意！
倪憲伯 ( Robert)

devvratsompura, 2017-06-02:
How to print over wifi? it is showing error. i have used mks-robin version (1.0.1) English.

closed 2020-04-18 by 3Dangel