makerbase-mks/MKS-Robin #3: USB Driver

Sensemaya, 2017-03-01:
Hello have the mks-Robin v 2 firmware
need the USB driver to conect Repetier host
Thanks

3Dangel, 2017-03-09:
Did you change baudrate in pc com port?

Sensemaya, 2017-03-09:
Hello Thanks for the reply
yes I did see the snapshot
The driver at github can not be installed by the installer in the zip file
make an error.
I install the driver looking by the device manager then pointing to the
file.
Nothing works I use Repetier Host to connect.
and will show is conected but unable to control the board, if the driver is
ok then the board is not working properly,
One of the problems that I see is than MKS do not provide with a Manual or
pinout of their product.
Thnsks to look for a solution.

Carlos

closed 2020-04-24 by makerbase-mks