makerbase-mks/MKS-Robin #7: Z- probe Auto level method

LSRoud, 2017-05-16:
line 98 bed level method changing the number does not change the ABL points it just keeps the 3 points which refers to the Z_PROBE_X1 to Z_PROBE_Y3 need grid bed leveling like delta becase 3 point doesn't compensate for the bend in metal plates witch are over a centain distance

problem with the z probe is the distance of the z probe hieght line 99 i need to be able to set the lines of code below because after homing than doing G29 gets error message that the z-probe does not untrigger meaning its to close to the bed but with proximity sensors they have to know the correct postion to the  nozzel so they work properly when auto bed leveling 

X_PROBE_OFFSET_FROM_EXTRUDER  
Y_PROBE_OFFSET_FROM_EXTRUDER 

 Z_RAISE_BEFORE_PROBING 
 Z_RAISE_BETWEEN_PROBING
 Z_RAISE_AFTER_PROBING

MD3DDelta, 2017-06-14:
Hi , LSRoud

Maybe you do like me , i have an delta printer .
first i do G32 I0.02 , 3 pionts leveling , then save ,M500.
then G29 , save again M500.
Set bed height  ,M306 Z0, save m500.
then without turning-off the printer And homing the printer
, start file that has to print. 

The printbed had to be straight as possible, if errors are more the 0.15mm , you allways have faills in the demenshings, most inside the prints.
code : 
G28 G32 I0.02
M500
G28 G1 X0 Y0 Z5 F5000 G29 G1 X0 Y0 Z50 F5000
M500
SET correct Height by jogging hotend,
M306 Z0
M500
G1 X0 Y0 Z50 F5000 = park position.
Start printing without G28.
Sorry for the bad english, is rusted .

cheers mike

LSRoud, 2017-06-14:
Sorry should of been more specific about the type of printer its not a delta its a corexy and my bed is 300x300 and i know its not flat

Danijele, 2018-05-22:
Hello together,
I have a Delta printer, unfortunatelly when I'm starting auto leveling than the probe crashes into the bed and is riddeleling over the top. I have no idea why this occours...here my configuration, may somebody support me in this topci  (I have an mechanical switsch installed 15,8mm underneeth the nozzle).
Thanks in advance Danijel
########### Auto Leveling ##############
>cfg_auto_leveling_cmd:G28;G32 S2;		#the setting of auto-leveling command button
>FEATURE_Z_PROBE			1		#0:disable leveling£¬1:enable leveling
>BED_LEVELING_METHOD		0		#0:3 points leveling,1:more points leveling,2: 4 points leveling
>Z_PROBE_ON_HIGH			1		#Probe signal(0: low level,always on; 1:high level,always off)
>Z_PROBE_HEIGHT				-15.8	#the height difference between Z-probe and  nozzle
>Z_PROBE_X1				-50			#coordinateX1 is preset point
>Z_PROBE_Y1				-50			#coordinateY1 is preset point
>Z_PROBE_X2				50			#coordinateX2 is preset point
>Z_PROBE_Y2				-50			#coordinateY2 is preset point
>Z_PROBE_X3				-50			#coordinateX3 is preset point
>Z_PROBE_Y3				50			#coordinateY3 is preset point
>cfg_leveling_z_speed			1.5		#the speed of Z moving when manual leveling(mm/min)
>cfg_leveling_xy_speed			5		#the speed of XY moving when manual leveling (mm/min)
>BED_LEVELING_GRID_SIZE 		5		#leveling interval
>Z_PROBE_SPEED				1.5		#the speed of Z-probe
>Z_PROBE_XY_SPEED			5		#the speed of XY