makerbase-mks/MKS-Robin #8: Printing over wifi!

devvratsompura, 2017-06-02:
how to print over wifi? In firmware  there are no settings for it.

themaxx2k, 2017-06-08:
Also want to know how to enable wifi module.

devvratsompura, 2017-06-09:
[MKS Robin Firmware.zip](https://github.com/makerbase-mks/MKS-Robin/files/1063810/MKS.Robin.Firmware.zip)

Just need to make adjustment in 'robin_config' for wifi settings as you require. And update through SD card as normal. after that connect to the board wifi. insert the ip from the display to pronterface!

themaxx2k, 2017-06-10:
is this suitable for board version 1 or 2 with eps wifi module?

devvratsompura, 2017-06-10:
Actually my printer is work in progress so i have not tested yet!

themaxx2k, 2017-06-10:
Ok. Where did you find this version? Which board are you going to use? Dont want to fry my new board with a false firmware flash... and i do not really know how or what i have to change to make it work... dont know which setting i have to change and how to bring it to my own network... is there a documented version of the config which includes wifi settings?

devvratsompura, 2017-06-11:
hi,
I am using mks robin controller board. I get this firmware form mks team.
And as i am knowing uploading the wrong firmware you can not fry the board.
"513151012@qq.com" this is mks team email address.

you just need edit the text file "mks_config" like your steps per mm, build
volume,No of extruder etc. Its easier then editing marlin. in this file you
will find settings for wifi too. Just make changes as you require.

Regards
Devvrat Sompura

.[image: Inline image 2][image: Inline image 1]

themaxx2k, 2017-06-11:
will have a deeper look into. are there any infos on which fw this version is based?

devvratsompura, 2017-06-12:
Not in english

themaxx2k, 2017-06-13:
Ok. The basic config is quite simple and connection via pronterface works well. Firmware is 2.04 now i will try to install it to one of my printers. Will report if it works

3Dangel, 2017-06-13:
devvratsompura
Which version of Tobin board do you have?
Cause I have v1.0 & V 2.1

devvratsompura, 2017-06-14:
2.1

themaxx2k, 2017-06-14:
Myself installed the above linked version on Robin 2.2. Update and basic function seem to be ok. not tested in the wild :-)

themaxx2k, 2017-06-14:
to get connected you have to enter the ip:8080 in pronterface and use 250000baud

devvratsompura, 2017-06-15:
Leave some photos of your printer when you done!

devvratsompura, 2017-06-17:
can you provide me your personal email?

themaxx2k, 2017-06-19:
Firmware and wifi is working. Already got it printing. Just adapted the settings from my repetier to robin. Works good.

yummyburger, 2017-09-14:
appreciate if anyone can help me with setting up wifi on prontoface please. my email is darius_k@hotmail.com. thank you

yummyburger, 2017-09-16:
managed to connect after 3 days of frustration with no help from Makerbase. Anyone intending to buy makerbase board, better think through. The english manual is so old, only the chinese manual is updated.

If you want to connect the wifi to your home network, easiest way is to change the settings in config file to

STA or AP : Choose STA
Network name change to your network
Password : Your network password and thats all

Why it took me so many days? The bloody wifi module cannot connect to 5GHZ band,so connect it to your 2.4ghz band network.

You can also add in the following to your config file

#baud rate setting(1:9600;2:57600;3:115200;4:250000)
>cfg_baud_rate:4

If you want to connect wirelessly to Prontoface

At the Port selection, just click and keyp : your ip address as shown on your touch screen , example 190.168.0.6:8080 and add 8080 behind it and off you go.

Then the baud rate select 25,000 and you are sorted.

Cheers

prasarnh, 2017-10-02:
hi every one my mks robin board is V2.3 so which firmware to use ,i just down load and save to sd when it reboot it show screen wifi updating for new firmware and wait for long time and show error message on top

emreyz, 2018-01-21:
Hello everyone could any body sent to me corexy parameters for 20x20x20 printer. I think I have some config problems I want to compare with another configuration my printer type is hbot.
Please be sent to me emreyz@hotmail.com.tr

#-------------------------------------------------------------
############ Configuration on Machine #######################
#machine type
#0:xyz
#1:corexy(x_motor = x+y, y_motor = x-y),
#2:corexy(x_motor = x+y, y_motor = y-x),
#3:delta

>DRIVE_SYSTEM 		        2				#machine type

>cfg_language_type		3				#language setting(Simplified Chinese: 1,  Traditional Chinese: 2, English: 3)
>cfg_ui_set_maxtemp  		260				#just for UI display;the max target temp setting of extruder,should be less than MAXTEMP
>cfg_ui_set_heated_bed_temp  	120				#ust for UI display;the max target temp of heated bed, should be less than HEATED_BED_MAX_TEMP

>NUM_EXTRUDER 			1				#number of extruder number(1:singal£»2:dual)
>HAVE_HEATED_BED 		1				#1£ºenable heated bed  ,0£ºdisable heated bed
>EXT0_TEMPSENSOR_TYPE 		1				#1:100k thermistor£»102£ºMAX31855thermocouple
>EXT1_TEMPSENSOR_TYPE 		1				#1:100k thermistor£»102£ºMAX31855thermocouple
>HEATED_BED_SENSOR_TYPE 	1				#1:100k thermistor£»

#-------------------------------------------------------------
############ Fuction settings #######################
>INVERT_X_DIR			0				#X motor direction, 1goes opposite direction;
>INVERT_Y_DIR			0				#Y motor direction, 1goes opposite direction;	
>INVERT_Z_DIR			0				#Z motor direction, 1goes opposite direction;	
>EXT0_INVERSE 			1				#E0motor direction, 1goes opposite direction;
>EXT1_INVERSE 			0				#E1motor direction, 1goes opposite direction;

>XAXIS_STEPS_PER_MM		320				#X steps per mm 
>YAXIS_STEPS_PER_MM		320				#Y steps per mm 
>ZAXIS_STEPS_PER_MM		820				#Z steps per mm 
>EXT0_STEPS_PER_MM		93				#E0 steps per mm 
>EXT1_STEPS_PER_MM		100				#E1 steps per mm 

>X_MAX_LENGTH 			200				#the MAX X-axis distance
>Y_MAX_LENGTH 			200				#the MAX Y-axis distance
>Z_MAX_LENGTH 			200				#the MAX Z-axis distance
>X_MIN_POS 			0				#the MIN X-axis distance
>Y_MIN_POS 			0				#the MIN Y-axis distance
>Z_MIN_POS 			0				#the MIN Z-axis distance

>MIN_EXTRUDER_TEMP		175				#Extruder MIN TEMP,play a protective role
>MAX_EXTRUDER_TEMP 		275				#Extruder MAX TEMP,play a protective role
>MAX_HEATED_BED_TEMP 		150				#Heated bed MAX TEMP£¬play a protective role

>HOMING_ORDER			1				#Set direction of endstops when homing;      
>X_HOME_DIR			-1				#Homing direction(-1:MIN£¬1:MAX)
>Y_HOME_DIR			-1				#Homing direction(-1:MIN£¬1:MAX)
>Z_HOME_DIR			-1				#Homing direction(-1:MIN£¬1:MAX)

>HOMING_FEEDRATE_X 		30				#the feedrate when X homes
>HOMING_FEEDRATE_Y 		30				#the feedrate when Y homes
>HOMING_FEEDRATE_Z 		5				#the feedrate when Z homes

# 1 means endstop always-on,0 is always-off
>ENDSTOP_X_MIN_INVERTING 	1				
>ENDSTOP_Y_MIN_INVERTING 	1				
>ENDSTOP_Z_MIN_INVERTING 	1				
>ENDSTOP_X_MAX_INVERTING 	0				
>ENDSTOP_Y_MAX_INVERTING 	0				
>ENDSTOP_Z_MAX_INVERTING 	0				

# 1 for Min/Max endstop enable in hardware,while 0 disable 
>MIN_HARDWARE_ENDSTOP_X 	1				
>MIN_HARDWARE_ENDSTOP_Y 	1				
>MIN_HARDWARE_ENDSTOP_Z 	1				
>MAX_HARDWARE_ENDSTOP_X 	0				
[... 165 lines cut ...]
>morefunc3_cmd:G28;
>morefunc4_cmd:G28;
>morefunc5_cmd:G28;
>morefunc6_cmd:G28;

#-------------------------------------------------------------
######  Display customization  #######

>cfg_background_color		0x494949			#the color of screen background

>cfg_title_color		0xFFFFFF			#the color of title text

>cfg_state_background_color	0x494949			#the color of temperature,fan,(etc)                                                                  background
            
>cfg_state_text_color		0xFFFFFF			#the color of temprature,fan,(etc) text

>cfg_filename_background_color	0x494949			#the color of file button

>cfg_filename_color		0xFFFFFF			#the color of file text

>cfg_printingstat_word_background_color		0x494949	#the color of background in printing status

>cfg_printingstat_word_color			0xFFFFFF	#the color of text in printing status

>cfg_BUTTON_3D_effects		0				#3D effects setting (disable: 0, enable: 1)

wedgevic, 2019-08-10:
how do I get the mks_config file? (sorry, Im new)