makerbase-mks/MKS-Robin #9: Jerk settings and autoleveling

jtingt, 2017-06-13:
I was wondering if there is any way of adjusting the jerk settings, the defualt seems too high. 
and has anyone figured out how to get more than three point autoleveling?

MD3DDelta, 2017-06-13:
Hi, g29 makes more pionts for leveling , but if you want info about the software go to smoothyware,
there you find the answers fast.
Just keep in mind that you have to controle out/input pins. MKS is some pins differend, the rest is same.
Link:
http://smoothieware.org/3D-printer-guide#your-guide-to-installing-smoothieboard-in-a-3d-printer

cheers mike

3Dangel, 2017-06-13:
Md3dDelta
Hey Mike
I think you have mistaken Robin board with the SBase!
Unless you are telling us that this Robin is using smoothie!?

Please confirm 
Thanks
Bob

MD3DDelta, 2017-06-14:
Hey Bob , 
i tought it was also like mks sbase , smoothieware, i see only  for the tft it is.
I have only experience with marlin and smoothieware for delta printers.
I have no robin board , 
I have set the" segments per mm " in 200 and not the default 100
the jerk was 10 set to 5, and all prints went super smooth out.
but this is for delta ,others try and you see, thats how i do it.

if you can give the config-file, i can look to it , maybe i find an solution , 

Hope you get it right . 
cheers mike

3Dangel, 2017-06-14:
:-) hi Mike,
Taking the time to share it is already a big thing!! Judi for what you are doing!!!
The Robin is a STM 32 bit board that is CLOSED source FW, so we don't know much of it, beside of thinking it is well a Repetier FW that is compiled and released whenever MKS team do. 

Differently of SBase, it is a clone from the open source Smoothieware, the only open source compatible clone board out there. 

I hope your info could serve as pointer to others, since all printers FW share some thing alike in them!

Cheers

Bob

3Dangel, 2017-06-14:
I mean kudo not Judi !!