makerbase-mks/MKS-Robin #10: Upload and print from SD failed

themaxx2k, 2017-06-22:
I tried the Robin Fw 2.03 an 2.04 on Robin 2.2 Board, but i am not able to upload G-code to SD-Card, also i am not able to print g-code, stored on the SD Card. Printing over Wifi works good, printing with Octoprint produced some comms-failures (maybe i Need to reroute the usb-cable...). What are legit filemames an extention of the gcode-file? which type is supported? SD, SDHC... and which filesystem-type works best? FAT16, FAT32...

themaxx2k, 2017-06-25:
Ok. Printing works when formatin the SD to FAT16 and partition smaller 2GB (not tested more combinations) but upload (repetier or pronterface) does not work. always displays transfer failed.
And strange rattling sound on higher speeds (33mm/s and up) with DRV8825...

Elnurjin, 2017-06-26:
Hello. Help please solve the problem. I upload the code to the flashcard. Then I press the seal. But the seal does not start, the nozzle and the table not hot. Why? =(

themaxx2k, 2017-06-27:
mine starts printing when i resized the Partition to under 2GB and used fat16 as filesystem. now printing works (but i also got one break during print, dunno why, maybe due to to many connected users or USB-cable). now i suffer with odd rippling Sound on higher Speeds....

Elnurjin, 2017-06-27:
I solved the problem. Robin can not start if there are comments in the code with the content of the Cyrillic.

Elnurjin, 2017-06-27:
Now I have a different problem. I make a speed of 70mm \ in the slicer, but the printer prints at a higher speed. Why is this happening?