makerbase-mks/MKS-Robin #11: customised graphics

berezkinav, 2017-07-05:
Which setting should i use (in Image2Lcd) to create my own logo? How can i convert *bmp.bin files back to images, or are there somwhere these images as bitmaps in the firmware folders? I wish to produce printers with your board, and want to have a full control over the user interface (logo and buttons).

makerbase-mks, 2017-07-31:
You can save your logo image in BMP format,and then convert the software into.bin file via Image2Lcd.Finaly,use SD card to update your logo.

makerbase-mks, 2017-07-31:
Do you have an email account? I can send you how to configure Image2Lcd.

pirogeth, 2017-10-02:
Why can't you upload a pdf to show how to configure it so everyone can see it?

csorrows, 2018-09-17:
I'm not going to post my email address publicly. Can you please provide a guide on how to modify the images? Even if it is just the Image2LCD settings, that would be great.