makerbase-mks/MKS-Robin #12: Config of 2 extruders problems

Elnurjin, 2017-07-28:
Help please adjust 2 extruders. I have hotend Chimera. The distance between the nozzles is 18mm. In the configuration I indicated 2 extruders. In the slicer indicated the distance between the nozzles 18mm. But the nozzle does not print when the offset is printed. I print a square with layers of different colors. And the printer prints 2 cubes separately.
What could be the problem?
[Photo print](https://pp.userapi.com/c639526/v639526211/38778/Tc0pf-dEuXw.jpg)
[This is a photo of a Gcode](https://pp.userapi.com/c639526/v639526282/36a77/WIWFWdpIJw4.jpg)
My start code:
M201 X1500 Y1500 Z100 E1500
G28 
G21 ;metric values
G90 ;absolute positioning
M107 ;start with the fan off
T1 ;Switch to the 2nd extruder
G92 E0 ;zero the extruded length
G1 F200 E10 ;extrude 10mm of feed stock
G92 E0 ;zero the extruded length again
G1 F200 ; 
T0 ;Switch to the first extruder
G92 E0 ;zero the extruded length
G1 F200 E10 ;extrude 10mm of feed stock
G92 E0 ;zero the extruded length again
G1 F200
M218 T0 X18 Y0;
M218 T1 X0 Y0;
M117 Printing...

G1 Z200 F3000
G1 X10 Y90 F3000
G1 Z0 F3000
G1 E3
G1 Y130 F3000
G1 F400 E2.45
G92 E0
G1 Z0.3
G92 Z0

makerbase-mks, 2017-07-29:
Hello,which product do you use?

Elnurjin, 2017-07-29:
MKS Robin (simplify3d)
The second problem is double the speed of printing. I make speed 50mm \ s
(slicer) but the printer prints about 100mm \ s

makerbase-mks, 2017-07-31:
Can you send me some photos?