makerbase-mks/MKS-Robin #15: BLTouch installation on Robin v2.2

Uglystick, 2017-08-27:
Could you please advise which pins on the board are used for the 3 wire connector on the BLTouch.

Could you also advise what setting on the firmware (robin_config.txt) need to be made for it to work correctly.

I have searched extensively and i have not been able to find any clear answers to this issue.

makerbase-mks, 2017-08-28:
The robin is not supported BLTouch. You can used Photoelectric switch or Proximity switch to replace it.

pirogeth, 2017-09-19:
I purchased your board and tft screen. the bl touch is a huge printer part now. It would be in your best interest to make this boards compatibility with bl touch. I'm holding off using your board because of this issue.

glatzy91, 2018-03-20:
Ist the Bltouch working in the latest release?

pirogeth, 2018-03-20:
No

closed 2018-05-29 by makerbase-mks

3D-Druckerzentrum, 2018-08-28:
I am also still waiting for the BLTouch to be supported by MKS Robin.
I have printers with the MKS Base supporting the BLtouch and other printers with the MKS Robin not supporting the BLTouch.
So I have to tell some of my customers that they don't have the possibility to upgrade their printer with a BLTouch sensor :-(