makerbase-mks/MKS-Robin #16: how to confirgure for cartesian printer

yummyburger, 2017-09-03:
how do i setup configure this board, after i flashed the firmware it just go haywire.

I am about to throw this board away, too bad i bought my printer with this in it and all i see everywhere is just delta including in the config file.

Can someone tell me how can i change it?

yummyburger, 2017-09-03:
most importantly, how can i set my print bed size

makerbase-mks, 2017-09-04:
If it's an XYZ machine type , the bed size is the value of your the MAX X-axis distance and MAX Y-axis distance.

yummyburger, 2017-09-04:
what about this , do i fill this in or leave it empty or just delete it?

#Delta parameters
>DELTA_MAX_RADIUS	135					#the radius of Delta annulus
>PRINTER_RADIUS		197					#the distance from machine center to vertical                                                                  top
>DELTA_DIAGONAL_ROD	346.75					#the length of Delta pole
>DELTA_FLOOR_SAFETY_MARGIN_MM	15				#the safe distance of leveling edge

makerbase-mks, 2017-09-08:
You don't have to worry about it, you just choose the machine type.
Like this:
">DRIVE_SYSTEM 		        0				#machine type"

yummyburger, 2017-09-08:
thank you. the printer manufacturer didn't give me the full config file so i don't know what needs to be in there. everytime they send me the file either is very short and every single time is different. 

Do i just download the 2.04, change the machine time to 0 and that will be ok and dont have to change the rest unless the motor is going wrong direction?

1 more problem, used to have E0 & E1 in the touch screen, but after recently with all the flashing. now it only shows E1 but my extruder is installed on E0 though, but it still work. Now the touch screen only show E1 and when i  try to print, after 10 minutes , it kept rebooting by itself.

yummyburger, 2017-09-08:
Here is my confirguration, can help me check please.

#-------------------------------------------------------------
############ Configuration on Machine #######################
#machine type
#0:xyz
#1:corexy(x_motor = x+y, y_motor = x-y),
#2:corexy(x_motor = x+y, y_motor = y-x),
#3:delta

>DRIVE_SYSTEM 		        0				#machine type

#baud rate setting(1:9600;2:57600;3:115200;4:250000)
>cfg_baud_rate:3

>cfg_language_type		3				#language setting(Simplified Chinese: 1,                                                                 Traditional Chinese: 2, English: 3)

>NUM_EXTRUDER 			1				#number of extruder number(1:singal£ª2:dual)
>HAVE_HEATED_BED 		1				#1£∫enable heated bed  ,0£∫disable heated bed
>EXT0_TEMPSENSOR_TYPE 		1				#1:100k thermistor£ª102£∫MAX31855thermocouple
>EXT1_TEMPSENSOR_TYPE 		1				#1:100k thermistor£ª102£∫MAX31855thermocouple
>HEATED_BED_SENSOR_TYPE 	1				#1:100k thermistor£ª

#-------------------------------------------------------------
############ Fuction settings #######################
>INVERT_X_DIR			1				#X motor direction, 1goes opposite direction;
>INVERT_Y_DIR			1				#Y motor direction, 1goes opposite direction;	
>INVERT_Z_DIR			1				#Z motor direction, 1goes opposite direction;	
>EXT0_INVERSE 			0				#E0motor direction, 1goes opposite direction;
>EXT1_INVERSE 			0				#E1motor direction, 1goes opposite direction;

#pause position (-1 is invalid; Z-axis is relative position)
>cfg_XPOS:-1
>cfg_YPOS:-1
>cfg_ZADD:10

>XAXIS_STEPS_PER_MM		80				#X steps per mm 
>YAXIS_STEPS_PER_MM		80				#Y steps per mm 
>ZAXIS_STEPS_PER_MM		1600				#Z steps per mm 
>EXT0_STEPS_PER_MM		94				#E0 steps per mm 
>EXT1_STEPS_PER_MM		94				#E1 steps per mm 

>X_MAX_LENGTH 			200				#the MAX X-axis distance
>Y_MAX_LENGTH 			200				#the MAX Y-axis distance
>Z_MAX_LENGTH 			200				#the MAX Z-axis distance
>X_MIN_POS 			0				#the MIN X-axis distance
>Y_MIN_POS 			0				#the MIN Y-axis distance
>Z_MIN_POS 			0				#the MIN Z-axis distance

>MIN_EXTRUDER_TEMP		150				#Extruder MIN TEMP,play a protective role
>MAX_EXTRUDER_TEMP 		280				#Extruder MAX TEMP,play a protective role
>MAX_HEATED_BED_TEMP 		150				#Heated bed MAX TEMP£¨play a protective role

>HOMING_ORDER			1				#Set direction of endstops when homing;      
>X_HOME_DIR			-1				#Homing direction(-1:MIN£¨1:MAX)
>Y_HOME_DIR			-1				#Homing direction(-1:MIN£¨1:MAX)
>Z_HOME_DIR			-1				#Homing direction(-1:MIN£¨1:MAX)

>HOMING_FEEDRATE_X 		40				#the feedrate when X homes
>HOMING_FEEDRATE_Y 		40				#the feedrate when Y homes
>HOMING_FEEDRATE_Z 		5				#the feedrate when Z homes

# 1 means endstop always-on,0 is always-off
>ENDSTOP_X_MIN_INVERTING 	1				
>ENDSTOP_Y_MIN_INVERTING 	1				
>ENDSTOP_Z_MIN_INVERTING 	1				
>ENDSTOP_X_MAX_INVERTING 	0				
>ENDSTOP_Y_MAX_INVERTING 	0				
>ENDSTOP_Z_MAX_INVERTING 	0				

[... 173 lines cut ...]
>morefunc3_cmd:G28;
>morefunc4_cmd:G28;
>morefunc5_cmd:G28;
>morefunc6_cmd:G28;

#-------------------------------------------------------------
######  Display customization  #######

>cfg_background_color		0x000000			#the color of screen background

>cfg_title_color		0xFFFFFF			#the color of title text

>cfg_state_background_color	0x000000			#the color of temperature,fan,(etc)                                                                  background
            
>cfg_state_text_color		0xFFFFFF			#the color of temprature,fan,(etc) text

>cfg_filename_background_color	0x000000			#the color of file button

>cfg_filename_color		0xFFFFFF			#the color of file text

>cfg_printingstat_word_background_color		0x000000	#the color of background in printing status

>cfg_printingstat_word_color			0xFFFFFF	#the color of text in printing status

>cfg_BUTTON_3D_effects		0				#3D effects setting (disable: 0, enable: 1)

valentinoo0, 2018-02-17:
Someone can tell me how to deactivate the pid of the heating bed to be able to use a solid state relay.
Thank you

closed 2018-05-29 by makerbase-mks