makerbase-mks/MKS-Robin #17: Any chance of open sourcing?

fonix232, 2017-09-18:
The Robin seems to be an awesome little board for controlling printers, however because it isn't open source, I'm more inclined to buy the SBase 1.3 instead.

If you were to release additional sources for compiling Smoothieware for the Robin, it would definitely make it a better selling product!

makerbase-mks, 2017-09-23:
Thanks for your advice,we will consider it.

mortrem, 2018-05-24:
It is sad: the lack of open source truly limite is application. I wish to use a laser with my mks robin but since I can't access the pinout... I'm stump. You should also realease the g-code of your board so we could optimized slicing or use other option Like the servo pin

3Dangel, 2018-05-24:
The flavour of G code should be Repetier.
That is after trying a lot of codes and also if you are familiar with
Repetier FW, you would know that the config are almost exact the same.

Hope this helps some of you.

Cheers

Robert

closed 2018-05-25 by makerbase-mks

mortrem, 2018-05-25:
While I'm aware of the similarity, confirmation would be nice. Or at least some kind of official info. Right now, unless you build a classical 3d printer, you stuck with a board that... Is very smart designed but not polyvalent

fonix232, 2018-06-01:
@makerbase-mks why did you close this ticket without any notice or explanation? It's not exactly the best way to ensure customer trust....

mortrem, 2018-07-03:
Because they have no intention to do it open source and they made that very clear. Even if they use repetier as base code witch is open source and a direct insult to the coder.