makerbase-mks/MKS-Robin #18: instructions to set up for delta 3d printer.

tylerericksonid, 2017-09-25:
is there a set of instructions I can use to configure the board to my particular printer

marcel2008, 2018-01-25:
No comment from MKS? Just purchased this board becaus I thought it was a fair price for a 32bit board. But the lack of support from MKS is just very disapointing! Is there anybody who would offer ther config for a kossel mini? If I wont find anything I will return the board to MKS

tylerericksonid, 2018-01-26:
Hey I have a kossel and have it figured out for mine I have the same lack of service. I have the anycubic pulley. If you want to config email me at te285@hotmail.com I will sent you a working configuration the will at least get you started.

marcel2008, 2018-01-26:
That would be great! Can you paste it here in the comments? I'm sure I won't be the last one having those issues! So others would surely appreciate it as well! Thank you

makerbase-mks, 2018-01-30:
#-------------------------------------------------------------
############ Based Settings( Required) #######################
#machine type
#0:xyz
#1:corexy(x_motor = x+y, y_motor = x-y),
#2:corexy(x_motor = x+y, y_motor = y-x),
#3:delta

>DRIVE_SYSTEM 		        3				#machine type

>cfg_multiple_language		1				#multi-language(enable:1, disable:0)

>cfg_language_type		3				#languages setting,this configuration is valid when "cfg_multiple_language" is disabled. 
								#(simplified Chinese:1; traditional Chinese:2; English:3; Russian:4; Spanish:5;French:6;Italian:7).

>NUM_EXTRUDER 			1				#number of extruder (1:singal；2:dual)
>HAVE_HEATED_BED 		1				#1:enable bed;  0:disable bed
>EXT0_TEMPSENSOR_TYPE 		1				#1:100k thermistor；102: MAX31855 thermocouple
>EXT1_TEMPSENSOR_TYPE 		1				#1:100k thermistor；102: MAX31855 thermocouple
>HEATED_BED_SENSOR_TYPE 	1				#1:100k thermistor；

>FEATURE_TWO_XSTEPPER		0				#change E1 singal into X , then dual X is available(enable:1; disable: 0)
>FEATURE_TWO_YSTEPPER		0				#change E1 singal into Y , then dual Y is available(enable:1; disable: 0)
>FEATURE_TWO_ZSTEPPER		0				#change E1 singal into Z , then dual Z is available(enable:1; disable: 0)

#---------------------------------------------------------------
################# Machine settings ############################
>EXT1_X_OFFSET			0				#E1 offset from the origin of X axis (mm)
>EXT1_Y_OFFSET			0				#E1 offset from the origin of Y axis (mm)

>INVERT_X_DIR			0				#X motor direction, 1 goes opposite direction;
>INVERT_Y_DIR			0				#Y motor direction, 1 goes opposite direction;	
>INVERT_Z_DIR			0				#Z motor direction, 1 goes opposite direction;	
>EXT0_INVERSE 			1				#E0motor direction, 1 goes opposite direction;
>EXT1_INVERSE 			0				#E1motor direction, 1 goes opposite direction;

>XAXIS_STEPS_PER_MM		100				#X steps per mm 
>YAXIS_STEPS_PER_MM		100				#Y steps per mm 
>ZAXIS_STEPS_PER_MM		100				#Z steps per mm 
>EXT0_STEPS_PER_MM		100				#E0 steps per mm 
>EXT1_STEPS_PER_MM		100				#E1 steps per mm 

>X_MAX_LENGTH 			125				#the MAX X-axis distance
>Y_MAX_LENGTH 			125				#the MAX Y-axis distance
>Z_MAX_LENGTH 			300.5				#the MAX Z-axis distance
>X_MIN_POS 			-125				#the MIN X-axis distance
>Y_MIN_POS 			-125				#the MIN Y-axis distance
>Z_MIN_POS 			0				#the MIN Z-axis distance

>MIN_EXTRUDER_TEMP		175				#MIN TEMP on Extruder ,play a protective role
>MAX_EXTRUDER_TEMP 		275				#MAX TEMP on Extruder ,play a protective role
>MAX_HEATED_BED_TEMP 		150				#MAX TEMP on heated bed ，play a protective role

>HOMING_ORDER			1				#Set direction of endstops when homing;      
>X_HOME_DIR			-1				#Homing direction(-1:MIN，1:MAX)
>Y_HOME_DIR			-1				#Homing direction(-1:MIN，1:MAX)
>Z_HOME_DIR			-1				#Homing direction(-1:MIN，1:MAX)

>HOMING_FEEDRATE_X 		40				#the feedrate on X homing
>HOMING_FEEDRATE_Y 		40				#the feedrate on Y homing
>HOMING_FEEDRATE_Z 		10				#the feedrate on Z homing

# 1 means endstop always-on,0 is always-off
>ENDSTOP_X_MIN_INVERTING 	1				
>ENDSTOP_Y_MIN_INVERTING 	1				
>ENDSTOP_Z_MIN_INVERTING 	1				
>ENDSTOP_X_MAX_INVERTING 	0				
>ENDSTOP_Y_MAX_INVERTING 	0				
>ENDSTOP_Z_MAX_INVERTING 	0				

[... 186 lines cut ...]
>morefunc3_cmd:G28;
>morefunc4_cmd:G28;
>morefunc5_cmd:G28;
>morefunc6_cmd:G28;

#----------------------------------------------------------------
##############  Display Customization  #######################

>cfg_background_color			0x000000		#the color of screen background

>cfg_title_color			0xFFFFFF		#the color of title text

>cfg_state_background_color		0x000000		#the color of temperature,fan,(etc)background
            
>cfg_state_text_color			0xFFFFFF		#the color of temprature,fan,(etc) text

>cfg_filename_background_color		0x000000		#the color of file button

>cfg_filename_color			0xFFFFFF		#the color of file text

>cfg_printingstat_word_background_color	0x000000		#the color of background in printing status

>cfg_printingstat_word_color		0xFFFFFF		#the color of text in printing status

>cfg_BUTTON_3D_effects		0				#3D effects setting (disable: 0, enable: 1)

makerbase-mks, 2018-01-30:
this is our config file for delta printer.

Danijele, 2018-05-22:
may you tell me where to connect the Z-Probe?`
Thanks 

Danijel

closed 2018-05-29 by makerbase-mks

miga66, 2018-12-09:
how chinge the saiz 200mm for 400mm form  mks robin v2.3