makerbase-mks/MKS-Robin #23: Autoleveling Core XY

Reinivann, 2017-11-28:
I would like to connect an autoleveling to a CoreXY(Tronxy X5S), At Home, the standard endstop stops responding and continues to move up because the sensor in home position XY has no contact with the bed. How can I now set the home position in the middle and not as programmed left front Z level.

sahory, 2018-03-23:
Hello there did you find any way to stop Z homing at aouto levleing ?

3Dangel, 2018-03-24:
You would probably need to play around in the config.
Home directions

You can invert them.
And establish your sensor in another places

sahory, 2018-03-24:
I pass this one i need to know how my auto levling say prope didnt untrigger pefor moving i didnt fined how to made it same as broping at smothi config can you help 😀😀