makerbase-mks/MKS-Robin #29: Disable PID Heated bed

valentinoo0, 2018-02-17:
Someone can tell me how to deactivate the pid of the heating bed to be able to use a solid state relay.
Thank you

framul, 2018-02-18:
You don't need to deactivate the PID if you do your temperature control will not work.
Your solid state relay will work fine with the PID control...

valentinoo0, 2018-02-18:
I had read what happens this board was necessary to deactivate the pid because otherwise the mosfet could be damaged

valentinoo0, 2018-02-18:
I could use Pronterface to calibrate the pid of the extruder?

framul, 2018-02-18:
If that is the case you should not use that device .. 
To have a temperature control you must have the PID working .. 
It is as simple as that .. if the mosfet cant handle the switching you should consider one that is able to handle the PID control .
It is  must to have the PID system.. 

😎@🥝🏠

3Dangel, 2018-02-18:
Yes, you can.
Take down what the PID shows and manually enter the P, I, D values to the
config file and rename the robin.bin and config. txt

Cheers
Bob

3Dangel, 2018-02-18:
I currently use SSR to run my hot bed
I’m using two power supplies.
Main pws to run the board => Extruder, hot stepper and from the main
board’s 12V hot end’s connector to the SSR
And the 2 nd pws runs through the SSR to the hot bed

framul, 2018-02-18:
Okay you are right all you need to do is rewrite the file .. 

I don’t understand why you do so difficult .. what is there is really good .
PS a relay that works properly if need the present one on the board handles day 10 amp 
The cost is low @ $ 10 dollars +/_ with Amazon or AliExpress .

I have build my own Machine and the given settings are fantastic and keeps things in an equal temperature.. 
😎@🥝🏠

framul, 2018-02-18:
you make it more complicated than need to be.. 

KISS 
Keep It Simple stupid 
😎@🥝🏠

framul, 2018-02-18:
Wow also to tricky to do any job… 

I have an MKS board and a simple display that can be purchased with the Board .. 

All I do is design any printed object with Fusion.. 
and convert the STL in a G-code via Slic3r .. a modern and reliable product.
Than move the file with all the temperatures setting from slic3r on a USB stick and shove it the Display ..
easy as is Mate … 
Pronterface is old and is bypassed with some more up to date stuff..

After all you want to print things and not to bothered with all those auto settings .. 
Believe me those auto settings are good ..
I print anything from PLA to ABS and Nylon all with heat control 
 

😎@🥝🏠

valentinoo0, 2018-02-18:
ok thank you very much to all, the truth is that I am new with this board, since I had always used open-source, but the truth seems that here we have people willing to help

framul, 2018-02-18:
This board is designed in the USA by the Smoothy board guys.. 
Designed as a open source board .. it is the most simplest board ever .. and open source ! 
It can be copied and made by every one all the the diners want is a donation…. 

I use the 32Bit version because atmel 16bit versions are just to simple and slow… to do the task..
All operations can be controlled via an ASCii (Notepad) written file .
The MKS people make the Board cheap and good however made in China.. yet you will see quality there .. at a good price.

Now-a-days everyone wants to Hack and change, forgetting that most things are designed with a purpose to get the most out of a design…  However if you feel the need to Hack…. why are you not starting to design your own Circuits from scratch … 
write your own software .. you will see so much real fun that will give you… Same as building your own 3D Printer.

My advice keep it simple .. go whats available .. get the best for your price in mind.
You'll see your concept will be a winner .. because you are using designs that is designed by many… 
That is power of crowd development … NOT the Hacking…..  because that is frustration .. 
Honest I been there and done that .. and realised the reality… !
its the final results that matter … not the Hacking that makes a mess of someones concept… 

Cheers Mate…!
  
😎@🥝🏠

3Dangel, 2018-02-18:
:-)
We all have been there.
We help each other’s with the open source’s spiritt!!
Cheers
Bob

3Dangel, 2018-02-18:
Hi Frank

I think you are talking about the MKS-Sbase board.
The Robin is a different one, isn’t it?
Or did I miss something here!!?
Thanks

Bob

valentinoo0, 2018-02-18:
In your printers you have installed some autoleveling system with this board?

3Dangel, 2018-02-18:
I just finished my Bowden upgrade after one year ( been very lazy)
It is in my goal to implement photo sensors.
I was able to do it with Melzi V2.0 under Marlin fw.
I will post my finding once I have some time.
Cheers
Bob

framul, 2018-02-18:
My MKS Smoothy board V1.3 has the ability to have multi corner levelling .
The software needed is part of the associated LCD display board.  
However I have a an odd setup where the Hot plate (bed) is my Z axis.. and use 4 individual motors on each corner .. 
I build to get the ultimate in squareness as I use this machine for making specialised components used in the medical industry. 
I needed super small tolerances .
Nothing is the standard .. however If you are more the art type and make things not that precise you can use the auto levelling available in the Smoothy board. I have seen it working fine .. 

😎@🥝🏠

3Dangel, 2018-02-18:
Yes, I think the MKS-Sbase works with the auto leveling.
Sinc itnis based on Smoothie board.

We are here talking about the Mks- Robin which is “close source”. ( they
use Repetier FW)

Cheers

Bob

valentinoo0, 2018-02-19:
then the mks robin uses repeated firmware ??

valentinoo0, 2018-02-19:
Repetier

3Dangel, 2018-02-19:
It seemed like it is.
The config file is the same way.

3Dangel, 2018-02-19:
But we don’t know the pin out, so we cannot change a thing

valentinoo0, 2018-02-26:
what software do you use to do the slicing with this board?

Has anyone noticed improvement from one to another?
thanxs

valentinoo0, 2018-02-26:
someone uses repetier with slicer?
I want to use it but in the choice of slicer firmware I do not know which one to choose (malin, repetier, smoothieware), does anyone know which one I should select?

3Dangel, 2018-02-27:
Repetier and Cura and s3d
But the new version or Repetier is giving me errors still investigating if
USB issue or Win10 issue.
Baud rate 115200

It seems to be Repetier FW but I could be wrong.

Cheers
Bob

valentinoo0, 2018-02-27:
but when configuring slicer the software for which the g-code is going to be generated, which one do you have? (repetier, marlin ...)

3Dangel, 2018-02-27:
I think it is Repetier.
Man, it is try on error.
I use the g code Command list found from wiki and tried.
Some works some won’t and be elimination you get the most suitable one.