makerbase-mks/MKS-Robin #32: Extrouder motor not working

sahory, 2018-03-16:
Hello there 
I have new mks robin 2.3 board i have E0 and E1 for extrouder non of them work i change the config for E1 to work as dule motor with X and its work good i use fram work 2.06 and the older framwark not working on my tft this is the only one work at my tft i think its an framwark problem but how to solve it ?

pirogeth, 2018-03-17:
If you're still in the refund zone try to fet a refund and get the ledge board. It's better customizable than this board and they have better online e support versus the no customer support here.

sahory, 2018-03-17:
Lol this will be nice solution if they didnt answer or give any support

pirogeth, 2018-03-17:
I will be avoiding mks in the future if I can at all avoid it.

3Dangel, 2018-03-17:
It all depends how you wanna to use a board
Mine is the 1.0 I believe, and still use it for my prisa i4 clone.  Updated
from Melzi V2

It will not be comparable to the open sources but for the price, for a
32bit board, it works for me

Printing from SD or Astroprint

sahory, 2018-03-18:
I have mks sbase 1.3 its workibg fine and i still use it but about this board extrouder wont move what ever i try 
So i need mks ppl to answer me and check the framwark for me a gain my robin is 2.3

makerbase-mks, 2018-03-21:
@sahory we will fix the bug as soon as possible

sahory, 2018-03-21:
hello there and thx for reply my config file is
#-------------------------------------------------------------
############ Based Settings( Required) #######################
#machine type
#0:xyz
#1:corexy(x_motor = x+y, y_motor = x-y),
#2:corexy(x_motor = x+y, y_motor = y-x),
#3:delta

>DRIVE_SYSTEM 		        1				#machine type

>cfg_multiple_language		1				#multi-language(enable:1, disable:0)

>cfg_language_type		3				#languages setting,this configuration is valid when "cfg_multiple_language" is disabled. 
								#(simplified Chinese:1; traditional Chinese:2; English:3; Russian:4; Spanish:5;French:6;Italian:7).

>NUM_EXTRUDER 			2				#number of extruder (1:singal£»2:dual)
>HAVE_HEATED_BED 		1				#1:enable bed;  0:disable bed
>EXT0_TEMPSENSOR_TYPE 		1				#1:100k thermistor£»102: MAX31855 thermocouple
>EXT1_TEMPSENSOR_TYPE 		1				#1:100k thermistor£»102: MAX31855 thermocouple
>HEATED_BED_SENSOR_TYPE 	1				#1:100k thermistor£»

>FEATURE_TWO_XSTEPPER		0				#change E1 singal into X , then dual X is available(enable:1; disable: 0)
>FEATURE_TWO_YSTEPPER		0				#change E1 singal into Y , then dual Y is available(enable:1; disable: 0)
>FEATURE_TWO_ZSTEPPER		0				#change E1 singal into Z , then dual Z is available(enable:1; disable: 0)

#---------------------------------------------------------------
################# Machine settings ############################
>EXT1_X_OFFSET			0				#E1 offset from the origin of X axis (mm)
>EXT1_Y_OFFSET			0				#E1 offset from the origin of Y axis (mm)
>INVERT_X_DIR			1				#X motor direction, 1 goes opposite direction;
>INVERT_Y_DIR			1				#Y motor direction, 1 goes opposite direction;	
>INVERT_Z_DIR			0				#Z motor direction, 1 goes opposite direction;	
>EXT0_INVERSE 			1				#E0motor direction, 1 goes opposite direction;
>EXT1_INVERSE 			0				#E1motor direction, 1 goes opposite direction;

>XAXIS_STEPS_PER_MM		100				#X steps per mm 
>YAXIS_STEPS_PER_MM		100				#Y steps per mm 
>ZAXIS_STEPS_PER_MM		400				#Z steps per mm 
>EXT0_STEPS_PER_MM		100				#E0 steps per mm 
>EXT1_STEPS_PER_MM		100				#E1 steps per mm 

>X_MAX_LENGTH 			200				#the MAX X-axis distance
>Y_MAX_LENGTH 			200				#the MAX Y-axis distance
>Z_MAX_LENGTH 			300				#the MAX Z-axis distance
>X_MIN_POS 			0				#the MIN X-axis distance
>Y_MIN_POS 			0				#the MIN Y-axis distance
>Z_MIN_POS 			0				#the MIN Z-axis distance

>MIN_EXTRUDER_TEMP		135				#MIN TEMP on Extruder ,play a protective role
>MAX_EXTRUDER_TEMP 		265				#MAX TEMP on Extruder ,play a protective role
>MAX_HEATED_BED_TEMP 		120				#MAX TEMP on heated bed £¬play a protective role

>HOMING_ORDER			1				#Set direction of endstops when homing;      
>X_HOME_DIR			-1				#Homing direction(-1:MIN£¬1:MAX)
>Y_HOME_DIR			-1				#Homing direction(-1:MIN£¬1:MAX)
>Z_HOME_DIR			-1				#Homing direction(-1:MIN£¬1:MAX)

>HOMING_FEEDRATE_X 		40				#the feedrate on X homing
>HOMING_FEEDRATE_Y 		40				#the feedrate on Y homing
>HOMING_FEEDRATE_Z 		10				#the feedrate on Z homing

# 1 means endstop always-on,0 is always-off
>ENDSTOP_X_MIN_INVERTING 	1				
>ENDSTOP_Y_MIN_INVERTING 	1				
>ENDSTOP_Z_MIN_INVERTING 	1				
>ENDSTOP_X_MAX_INVERTING 	0				
>ENDSTOP_Y_MAX_INVERTING 	0				
>ENDSTOP_Z_MAX_INVERTING 	0				

[... 186 lines cut ...]
>morefunc3_cmd:G28;
>morefunc4_cmd:G28;
>morefunc5_cmd:G28;
>morefunc6_cmd:G28;

#----------------------------------------------------------------
##############  Display Customization  #######################

>cfg_background_color			0x494949		#the color of screen background

>cfg_title_color			0xFFFFFF		#the color of title text

>cfg_state_background_color		0x494949		#the color of temperature,fan,(etc)background
            
>cfg_state_text_color			0xFFFFFF		#the color of temprature,fan,(etc) text

>cfg_filename_background_color		0x494949		#the color of file button

>cfg_filename_color			0xFFFFFF		#the color of file text

>cfg_printingstat_word_background_color	0x494949		#the color of background in printing status

>cfg_printingstat_word_color		0xFFFFFF		#the color of text in printing status

>cfg_BUTTON_3D_effects		0				#3D effects setting (disable: 0, enable: 1)

sahory, 2018-03-21:
MKS_ROBIN_V2.0.6_WIN8.zip is my firmware

sahory, 2018-03-24:
Hello 
The problem was it should be at minimum tempretuer to start working i will close this as the solve of my problem 
Regard all

closed 2018-03-24 by sahory