makerbase-mks/MKS-Robin #34: Board broken

pirogeth, 2018-03-21:
During use my screen flipped and went backwards. its off skewed as well. Reflashing the firmware only caused the machine to brick. How can i fix this?

makerbase-mks, 2018-03-31:
Maybe you should contact our sales representative for replacement

pirogeth, 2018-03-31:
I have. They told me to reflash it. I told them it doesn't work and what can we do next. They haven't responded since. I'd like to have a replacement.

makerbase-mks, 2018-04-08:
what's version of your tft32 firmware and hardware.

janeyam, 2018-04-14:
我也遭遇一样的问题，但是更换sd卡之后就可以完成固件更新，所以跟主板无关，是sd卡质量太差。

janeyam, 2018-04-14:
I also encountered the same problem, but after the replacement of the sd card can complete the firmware update, so nothing to do with the motherboard, sd card quality is too bad. translation software! !

closed 2018-05-25 by makerbase-mks