makerbase-mks/MKS-Robin #36: Burned ARM Chip Replacement

mucitteam, 2018-04-08:
Our "ARM STM32F103ZET6" chip on MKS RobinV2.3 board is burnt.
We'd like to replace it. We've found a batch with same code at the page > [AliExpress](https://tr.aliexpress.com/item/STM32F103ZET6-32-bit-Microcontroller-CORTEXM3-512K-Flash-Microcontroller-Chip/32823643892.html?spm=a2g10.search0104.3.63.560f57f1sYPtI9&ws_ab_test=searchweb0_0,searchweb201602_1_10152_10151_10065_10344_10068_10342_10343_5722611_10340_10341_10697_10696_5722911_5722811_10084_5722711_10083_10618_10307_10059_10534_308_100031_10103_441_10624_10623_10622_10621_10620_5723011_5722511,searchweb201603_25,ppcSwitch_2&algo_expid=d5bab087-1bca-4d10-8ff3-79d1fa6460c0-9&algo_pvid=d5bab087-1bca-4d10-8ff3-79d1fa6460c0&priceBeautifyAB=0).

Would it work, are they all compatible? Is there any other detail we need to be aware of?

makerbase-mks, 2018-04-13:
Will not work.

mucitteam, 2018-04-13:
what is your suggestion to repair the damaged motherboard?
I think the biggest mistake is to choose makerbase brand.
my motherboard's chipset is burned and I need to repair it by changing it

batuhan22662, 2018-06-24:
Merhaba mucitteam rica etsem iletisime geçebilir misin? Bu konu hakkında danisacagim.

closed 2018-07-12 by makerbase-mks