makerbase-mks/MKS-Robin #41: Heated bed exceeded max temperature

washeree, 2018-05-21:
Hello, i just setup my robin v2.3 board and it gives Heated bed exceeded max temperature error. I changed termistor but still same error. 
Also i would like to ask how can i connect printer via usb to slicer host?
Thank you.
[robin_config.txt](https://github.com/makerbase-mks/MKS-Robin/files/2022380/robin_config.txt)

MD3DDelta, 2018-05-25:
Hi Washeree
In sofware config . did you check the pins from your board with the software?
Mks  is a clone from smoothie and co , so always check the pins in /out . find first an working config.
I do not have robin but the other base version ,is different .
slicer usb connection , see youtube or what else , for your specific slicer.
When printer runs then look for bltouch -facebook  there is more news.
cheers mike

makerbase-mks, 2018-05-29:
@washeree what is your software version?

washeree, 2018-05-29:
@makerbase-mks hello, my version is 2.6 and also i tried 2.4 but all same. When i changed termistor cables also it reads same value.

washeree, 2019-01-12:
Still having same issue about reading the bed temp. Its showing 230C.

makerbase-mks, 2019-03-14:
It shows that the temperature of the hot bed is 230. Is it under heating or normal temperature?
If 230 degrees is displayed at normal temperature, it may be a problem with the temperature measurement port, or a thermal problem, or the thermal type of the profile setting is incorrect.

washeree, 2019-03-15:
@makerbase-mks I tried 3 different types of thermistor, it showing 230degrees without heating. Extruder temp is normal also i switch it to the hotbed but still same. As all we know you dont have many option on firmware for thermistor. Looks STM32 issue to me.

makerbase-mks, 2019-03-21:
If the value of the test remains the same, the temperature measurement pin on the hot bed is burned out.

washeree, 2019-03-21:
So yes its stm pin burned out right? Because its connected to the micro controller i guess so for control the temp and read temp.

washeree, 2019-09-10:
@github051 it was the lastest version later on i installed Marlin but was still same. Then i changed the processor and now its ok on Marlin and Mks software.