makerbase-mks/MKS-Robin #42: Delta Printer with mechanical Probe

Danijele, 2018-05-21:
Hello together,
I bought the Anycubic Kossel Plus Delta printer and after the Trigorilla was broken I Bought the MKS Robin V2.4.
So far everything seems to be o.k. excluding Auto bed Leveling is not working at all.
I connected the Z-Probe (it's ad on with mechanical Endstop) to -Z Enstop conection!
Problem is: When I'm performing leveling the machine goes down and when touching the hot bed the 
the whole Head rattles over the hotbed top. Even when I'm triggering the probe nothing works.
I setet up the Zero in the Center of Hotbed, means wwhen mooving down the nozzle stops 1/10 of a mm before the hotbed. I setetup the paramete: "Z_PROBE_HEIGHT	-15.8".
My problem is that the Probe after touching did not rises up, goes to next probing coordinate and plunges down for next triggering......May somebody help me or rather send mey maybe the config file form Delta with working Autoleveling?!?

Thank you in advance!
Danijel

Danijele, 2018-05-23:
Hello togehter,
problem solved....If you need help just ask! So far MKS Robin is working propper with my Delta Printer!
Danijel.Timic@gmx.de

3Dangel, 2018-05-24:
What was your solution?
Thanks for sharing !!
Robert

3Dangel, 2018-05-26:
> Hello Robert,  first Off all technical parameters and especially z-length
> Probe z hight has to be accurate. Z Probe Connect to -Z Endstop ans thats
> it!
> Best wishes!
>
> ⁣Gesendet mit BlueMail ​

Thanks for the info.
Cheers
Ribert

closed 2018-05-29 by makerbase-mks