makerbase-mks/MKS-Robin #45: Corexy homemade

alvaromendes, 2018-06-12:
Good evening my friends, im Brazilian, so Im sorry about my bad english.
So... I will try to do a new 3d printer machine and I decided to do a corexy ass well. My question is someone have a profile to build one and configurate with mks robin? Im tring to use dual extrusion too. The problem is, i cant found no one information to do this, if somebody help me i will be thankfull.

mattdog01, 2018-08-23:
Olá Alvaro,
I'm Canadian, sorry about my bad English. :)  I have not been on GitHub for a while so I just saw your message now.

I have made a made a 500 x 500 x 500 corexy printer based on the Tech2C design (check it out on YouTube).  I have attached my config but I'm not too sure if it will help you unless you have built your machine the same way.  
The config file in the Robin folder is very easy to understand and very basic so it should be too hard for you to tweak it for your needs. This file just needs to be loaded onto an SD card, inserted and then start your MKS Robin.  The Robin will upload the new configuration and then change the name of the file so it does't use it again. Make sure that the config file is not in a folder.  You can have other files on the SD card but the config must not be in a folder.

[robin_config.txt](https://github.com/makerbase-mks/MKS-Robin/files/2315511/robin_config.txt)

I have included this picture so you can see what my printer looks like.  (I'm still putting it together but it is printing well.  I am in the process of installing 3 belt driven Z axis lead screws.  The big platform is hard to keep perfectly level.

[image]

I hope this helps.
Matt

pauloeduardogodoy, 2021-08-19:
Hi Matt! That was exactly what I was looking for!

I intend to do the same, but in my case I have an MKS Robin E3D... I belive it is more limited in resources... But for now I just stared to scratch some ideas in a piece of paper.

Thank you for your post!