makerbase-mks/MKS-Robin #46: mks robin v2.3 & corexy

rigimod, 2018-06-16:
Hello
I made a corexy machine with mks robin v2.3 card, but I am having problems installing the software. axes work like cartesian. what do I need to do step by step? thank you
[image]
[image]
[image]

3Dangel, 2018-06-17:
Rig mod,

You would need to go into the config.txt and change the machine type.
And play around.

There will be very little support from China.

Good luck.
Bob

makerbase-mks, 2018-07-12:
You can configure your machine by following the information in the configuration file.

closed 2018-07-17 by makerbase-mks