makerbase-mks/MKS-Robin #47: Lots of issues...

JJJpnt, 2018-07-03:
Hi

First, I want to say that I'm quite keen on your hardware, so you can understand why I'm disappointed and how I hope that I'm mistaken. 

So : I bought a brand new Robin from a friend who actually never used it. I understood why.
First absolutely no doc, and on top of that as it's closed source (I'll come back to the open/closed source aspect a little later), no way can I dig into the src to understand what goes wrong.

Bugs. Features are wayyyyy minimal. PID on bed ? Forget it. Leveling ? No probe offset, no mesh, more points algo hits the wall past endstops... Max power for bed ? Damn you're dreaming. NC endstops, ok, but for X you have to set it to NO in the config... And wait what... M503 is missing. List goes on (and on and on). OK ! Is thermal runaway protection and other securities even implemented ? How can I know ? Just makes me wonder if anyone actually uses a Robin.

Now opensource : I don't mind an attempt to do a closed source, I don't mind selling opensource stuff. But : Displaying the reprap logo at startup of a closed source thing is already strange. The "PID Autotune finished ! Place the Kp, Ki and Kd constants in the Configuration.h or EEPROM" message seems oddly familiar to my eyes too. 

Now don't hear me wrong : the hardware is neat, idea is great and I happily used several of your boards.
But, why ? How ?

JJJpnt, 2018-07-04:
Correction of my previous message :
About Bed PID : After trial and error, found out it's repetier syntax : so M303 P1 S[temp]

makerbase-mks, 2018-07-12:
Thank you for your question, robin's firmware is constantly being updated.

closed 2018-07-17 by makerbase-mks