makerbase-mks/MKS-Robin #52: Get marlin running in these if you really want to see them sell

zillarob, 2018-09-02:
You guys are smart enough to make some really neat products, but how on earth can you handicap this board by not making the most popular firmware used in this hobby work on it? 
This is insanity, and I can't even conjure up any logical reason why anybody would do this to their product.
Maybe other companies are paying you to keep it under wraps? I have a feeling these things would be flying off the shelves if Marlin worked on it, and that would take a major bite out of the other 32bit boards in the market.

Consider the Base and Gen models. These are some of the best and most convenient 8bit boards available. 
Do you think anybody would buy them if they couldn't flash whatever firmware they wanted on them? 
I sure don't.

Same holds true for the SBase and Robin.
Get some info out there and maybe even help the Marlin community get things up and running and you will have the 32bit board market nailed down. You would have all the options covered at an excellent price point, and I'd almost feel bad for the other mfg because they would lose so many sales.

In my case, I have a Base and a Gen already. I would like to check out the 32bit stuff. 
The SBase would cover that, but I dont want integrated drivers. 
This leaves the Robin, but I'm not going to wall myself into a corner by not having a common and open source firmware available for it.
You get Marlin running on this thing and you will have at least 1 more sale from me, and I have a feeling you would see many many others swarm to it also.

zillarob, 2018-09-05:
I dont see where the lite or mini fit in. They already have the SBase with integrated drivers, not sure why they would go that way with a version of Robin. I think they would have been better off keeping the 2 lines separate like Base (integrated drivers) and Gen (swappable drivers).

Not having swappable drivers is a no go for me.

zillarob, 2018-09-05:
Id be ok with them abandoning the lite and mini series, both SBase and Robin.
With the huge variety of pololu pattern driver options we have out nowadays, I can believe anybody would buy a board with integrated drivers in the 1st place.

pen25, 2018-09-05:
you can solder headers on and use external drivers.

zillarob, 2018-09-05:
Thats true. For cnc stuff that uses the bigger ext drivers anyway, that wouldnt be too bad.
But for what these boards would get used for a huge majority of the time, that extra bulk would make the mini or lite versions pointless.

zillarob, 2018-09-05:
I like most of their products. They seem to be good quality, intelligently designed, and at a good price point.
They really hit it out of the park with the Base and Gen lines. Nice tidy and convenient packages that follow a standard that allowed them to become wildly popular.

I just dont understand why they would go the route they have with the 32bit stuff. Its like they decided they want to be like Apple or something and went completely off the rails =) They would be absolutely killing it if they stuck to the same plan on the 32bit stuff as they did with the 8bit. 
I think they should have paralleled what they did with the 8bit stuff- SBase = integrated drivers like the Base, Robin = pololu pattern drivers like the Gen, and let out enough info so that the community can get the common firmwares running on it. Compatibility will open many, many doors here.

I remember the tail end of them trying to make the reprap community happy. I understand why they werent willing to do what they required, and why that may have left a bad taste in their mouths. Shoot, those guys were rabid all-or-nothing maniacs and left a bad taste in my mouth =) They need to accept that some info is better than none for the community. Throw the "not open source" nag label on it, but ffs, dont delete the info so nobody has access to it. While fully open would be nice, I can see why MKS cant release things like gerber files to cover their butts a lil.