makerbase-mks/MKS-Robin #53: Endstops don't work and homing position

vk220, 2018-09-02:
Hi, about endstops: I connect simple mechanical normally open endstops to pins "g" and "s", endstops located on min of x, y and z axes. Part of my config below, but when I move extruder in minimum direction, it try to move even when endstop switched on. Where I can find schematic for connecting these endstops?
```
# 1 means endstop always-on,0 is always-off
>ENDSTOP_X_MIN_INVERTING 	0				
>ENDSTOP_Y_MIN_INVERTING 	0				
>ENDSTOP_Z_MIN_INVERTING 	0				
>ENDSTOP_X_MAX_INVERTING 	0				
>ENDSTOP_Y_MAX_INVERTING 	0				
>ENDSTOP_Z_MAX_INVERTING 	0				

# 1 for Min/Max endstop enable in hardware,while 0 disable 
>MIN_HARDWARE_ENDSTOP_X 	1				
>MIN_HARDWARE_ENDSTOP_Y 	1				
>MIN_HARDWARE_ENDSTOP_Z 	1				
>MAX_HARDWARE_ENDSTOP_X 	0				
>MAX_HARDWARE_ENDSTOP_Y 	0				
>MAX_HARDWARE_ENDSTOP_Z 	0				

# 1 for Min/Max endstop enable in software,while 0 disable 
>min_software_endstop_x 	0				
>min_software_endstop_y 	0				
>min_software_endstop_z 	0				
>max_software_endstop_x 	1				
>max_software_endstop_y 	1				
>max_software_endstop_z 	1				
```
And about homing: is it possible to detect home position by endstops switching? At now, seems to be that home position is where extruder was at power on.
Thank you.

zillarob, 2018-09-02:
"endstop always-on,0 is always-off"
Guessing they mean normally open and normally closed by this??
If that is the case, what you are doing seems like it should be correct.

Due to the abundance of info available for this board, we get to guess.
I think I would unplug the endstop from the board and measure the voltage at the S pin to the G pin.
If you have 5v, you are doing it right and might just need to switch the top ones back to 1s.
If you have 0v, you might need to swap from the G pin to the V pin.

vk220, 2018-09-02:
> Guessing they mean normally open and normally closed by this??
Yes, guessed like that. But seems that this is not true. I found the chinese datasheet and it says (by google translator) that 1 means normally open.
Changed settings back to 1s and it supposed to work fine.
Also, voltage at the S pin to G pin is 3.2V, and at V to G is also 3.2V...
Thanks.

zillarob, 2018-09-02:
Sounds like you are on the right track.
If you got 3.2v S to G, they are prob using 3.3v logic instead of the 5v I was guessing, and you are seeing the pull up voltage.
Having that pull up means you have the endstop on the correct pins (S and G).