makerbase-mks/MKS-Robin #54: mks robin mini update problem

amirfpv, 2018-09-07:
Hello, i tried to update my mks robin mini to new 2.0.7 firmware, but after update process completed, the screen became black and "readyprint" message showed on the screen.
i emailed to mks support but no reply!!!
if somebody updated his robin mini to 2.0.7 successfully please let me know or upload the zip file here.
thank you

amirfpv, 2018-10-15:
Close source firmware without any support!!
Waste of money and time

amirfpv, 2018-10-17:
> Hello, what is the hardware version of your Robin mini? If you have a
> longer version of robin mini 1.0, you may not be able to update the
> firmware of 2.0.7.

Yes i have robin mini v1.0
So what should i do now? Auto levels doesn't work probably in  firmware
2.0.6

amirfpv, 2018-10-17:
> Yes, robin mini v1.0 can only be updated to 2.0.6

Yes what? I have problem with auto leveling and firmware 2.0.6 is not
complete
It seems mks sell it's prototype boards first and stop support after a
short time
Why mks mini v1.0 not updated to firmware 2.0.7 while it has problem with
firmware 2.0.6?!!!

stickfisch, 2019-02-02:
Its possible with robin mini v2 can input jerk settings?