makerbase-mks/MKS-Robin #60: Strange things after auto-leveling

vk220, 2018-10-13:
Hi! Tried different auto-leveling modes, issue is that after auto-leveling procedure the display shows only graphics, but not text and D7 LED (if I see it right, it is near USB-socket) blinking green. I can start print job, but cannot see filenames. Also pronterface cannot connect (by wi-fi), before auto-leveling connect is ok.
So, what could it be and maybe there is possible to set some logging for debug?
Thanks.
My config is attached
[robin_config.txt](https://github.com/makerbase-mks/MKS-Robin/files/2475690/robin_config.txt)

makerbase-mks, 2018-10-15:
Will it appear after automatic leveling? Can you show it, the display picture of the touch screen after it

vk220, 2018-10-15:
Exactly, so.
Before auto-leveling it looks like:
[image]
Then, Print menu and Heat menu:
[image]
[image]
Ok, I start auto-leveling, after it stops I see next:
[image]
[image]
[image]
But I steel can tap on the file (without the name) and printing will start.

makerbase-mks, 2018-10-16:
What is the hardware version of your motherboard and the version of the firmware?
If you don't use the win8 style interface, use the red or blue style interface, will the same situation occur?

vk220, 2018-10-16:
MKS Robin v 2.2, firmware MKS Robin v2.0.7
Didn't try red or blue interface, just tried to switch multilanguage mode on, but the same situation occur.
Will try another style interface later.

vk220, 2018-10-16:
Sorry, came home and checked again: MB ver 2.3. TFT ver 1.1R, firmware ver 2.0.7
Tried blue style interface, but situation just the same.
Seems like there is some software error, maybe something wrong in the config, but I just don't know where.

makerbase-mks, 2018-10-22:
Hello, can you please improve your problem? After the auto leveling is automatic, will the touch screen still display abnormally? Have you said that changing to other styles is also the interface? I have not seen the same problem for the test on my side. Do you have other modules besides the modules that are automatically leveled?
If it is still the same situation, I will reflect this problem with our R&D staff and find out the cause and fix him as soon as possible. For your convenience, send the firmware and configuration files you are using now, and test them here.

3Dangel, 2018-10-22:
你們好，
若貴公司可以錄制how-to 基本設定 及自動水平功能和所使用的硬體，這樣，可幫助不少使用貴公司的Robin 板 的客戶。

祝 你們大伙 萬事如意

Robert
Calgar, AB

vk220, 2018-10-22:
> Hello, can you please improve your problem?
Hi,
OK, I'll try, 
Some photo:
MB view:
[image]
Before autoleveling:
[image]
[image]
And right after auto-leveling:
[image]
[image]

As you see, I have changed interface to blue style, but it didn't work. Also I've tried to change auto-leveling method to 3-point method. But all the same. After all, I've tried to reload all firmwares, but got nothing.
Not sure exactly what you mean about other modules. I have only wi-fi module and drv8825 installed.
My firmwares and config file:
[robin_config.txt](https://github.com/makerbase-mks/MKS-Robin/files/2503319/robin_config.txt)
[Robin.zip](https://github.com/makerbase-mks/MKS-Robin/files/2503321/Robin.zip)
[MksWifi.zip](https://github.com/makerbase-mks/MKS-Robin/files/2503322/MksWifi.zip)

3Dangel, 2018-10-23:
One thing you are not mentioning is what type of sensor you are using for
your Z.
Also on the picture, you have Two red cable in. What are they?
Auto leveling is much “complicated”, just saying you have problem and with
screen shoots, I don’t thing anyone can help much.
Best is to upload a video and close up of your connections and sensors/
limit switches and perform your auto leveling and showing the issue you
have.
This way, the MKS team might have better chance to help you, I think.

Best of luck.
Robert
Calgary, AB

vk220, 2018-10-23:
Guess, you're right, thanks. Just forgot about sensors.
So, z-probe is a cheap sensor, [something like that](https://ru.aliexpress.com/item/3D-Z/32851902530.html?spm=a2g0v.search0104.3.207.4ecc22ba5NumRW&transAbTest=ae803_4&ws_ab_test=searchweb0_0%2Csearchweb201602_1_10065_10068_318_319_317_10696_450_10084_10083_10618_452_535_534_533_10307_532_5723511_10301_5728311_204_10059_10884_10889_323_10887_326_100031_320_321_322_10103_448_449-10889%2Csearchweb201603_35%2CppcSwitch_0&algo_pvid=8f696562-ea75-4e69-92c3-9b7dd97a36d0&algo_expid=8f696562-ea75-4e69-92c3-9b7dd97a36d0-30)
(I had overheat it accidently, and now it doesn't hold on the nozzle)
This probe connected in parallel to min z-endstop (that's why there are 2 red wires in). I cannot connect it without z-endstop because this probe should be removed before nozzle heat up.
Another endstops are X and Y minimum position. Homing performing well. Endstops - are simple micro switchers.
[Here's a video](https://youtu.be/annuoHaggWM)
Hope it will help.

3Dangel, 2018-10-24:
> (I had overheat it accidently, and now it doesn't hold on the nozzle, but
> still works)

I believe that is the problem.
Usually, tit is either an end stop , it an auto leveling.
Your sensor needs to have positive, negative (or ground) and SIG.

I think you would need to acquire a bit more about how to install auto
leveling on Marlin’s Prusa i3 platform and learn from there.

I used to have only one SIG cable connected to a compatible board and
taking the power from the same power supply ( black and red )

As for Mks-Robin,  I’m not sure if the design & engendering department will
have the time to post a solution.
But, I have hope that they will help at their time frame.

Good luck.

Robert

3Dangel, 2018-10-24:
here is from repetier's site info about the auto leveling.
https://www.repetier.com/documentation/repetier-firmware/z-probing/

I personally, stop using it, since I know how/when/ what to look for
leveling my bed at very begining of my print... Specially when the nozzle
lay down the skirt or the first layer..

cheers
Robert

vk220, 2018-10-24:
> Your sensor needs to have positive, negative (or ground) and SIG.

In my case it works like simple micro switch, it has open contacts, when nozzle press on it, contacts are closing. So, I use only SIG and GND. For first time I just removed z-endstop and used z-probe instead. And all was the same. But when I start print I should remove this sensor, and after nozzle heats up the printer moves nozzle to home position and some z-limiter is needed. So I've connected z-endstop again.

3Dangel, 2018-10-24:
I see...

So it has to do something in the TFT touch screen and the main board that
can not recall the file name and show it on screen.....

It seems to be the memory is getting max out or a conflict that after
you've done the auto-leveling, you can not see the on screen files'
name.....

have you tried out, with the SD card out first and do leveling. and then
insert the SD card?

I'm still running the Robin V1.0 with manual leveling...

hope you find the answer from MKS

3Dangel, 2018-10-24:
again .... I think the usage of the Bed-Leveling _ Method 2 should
accompany for points as:
>Z_PROBE_X1 50 #coordinateX1 is preset point
>Z_PROBE_Y1 90 #coordinateY1 is preset point
>Z_PROBE_X2 90 #coordinateX2 is preset point
>Z_PROBE_Y2 50 #coordinateY2 is preset point
>Z_PROBE_X3 90 #coordinateX3 is preset point
>Z_PROBE_Y3 180 #coordinateY3 is preset point
>Z_PROBE_X4 180 #coordinateX4 is preset point
>Z_PROBE_Y4 90 #coordinateY4 is preset point

your video shows it started from 50,50=>180.180=>180,50

and thereafter it hangs there....

give it a try the above value and let me know...
I'm very curious....

my thoughts and reasoning is that MKS is using the Repetier FW as the base
and the Bed_Leveling_method_2 would correspont to (see below)...

if my assumption is correct, your auto leveling couldn't complete and
stalled your system...... GOOD LUCK

BED_LEVELING_METHOD 0

This method measures at the 3 probe points and creates a plane through
these points. If you have
a really planar bed this gives the optimum result. The 3 points must not be
in one line and have
a long distance to increase numerical stability. Delta printers should have
them as close to the columns as possible.
BED_LEVELING_METHOD 1

This measures a grid. Probe point 1 is the origin and points 2 and 3 span a
grid. We measure
BED_LEVELING_GRID_SIZE points in each direction and compute a regression
plane through all
points. This gives a good overall plane if you have small bumps measuring
inaccuracies.
BED_LEVELING_METHOD 2

Bending correcting 4 point measurement. This is for cantilevered beds that
have the rotation axis
not at the side but inside the bed. Here we can assume no bending on the
axis and a symmetric
bending to both sides of the axis. So probe points 2 and 3 build the
symmetric axis and
point 1 is mirrored to 1m across the axis. Using the symmetry we then
remove the bending
from 1 and use that as plane.

vk220, 2018-10-24:
> have you tried out, with the SD card out first and do leveling. and then insert the SD card?

No, I don't. I think it's not an SD-card issue, because all text disappearing: button labels, text in the "About" menu and so on.

> and thereafter it hangs there....

I don't think it hangs, because I can start printing even if I cannot see filenames.
When the bed placed unplanar too much, I get the error "z-probe switches before probing". Then, nozzle returns slowly to start position, and touch screen doesn't work untill printer restart.

technik-gegg, 2018-10-25:
@vk220 
I agree. The culprit isn't the Z-probe at all. I'm having the same problem with the Robin. In fact, the filenames (and other stuff) don't disappear but the color changes to black font on black background.
If you try hard enough, you can see other colors as well (i.e. green background, black font) which leads to only one conclusion:
The Z-Probe measurement somehow messes up the internal memory, in which the color codes are stored (a typical problem, when data is stored in an array or class and the data exceeds assumed boundaries).
This might be the tip of an iceberg, since depending on the length of the data stored, it could affect other parameters as well, which could result in the printer going nuts.

I've tried various combinations too (different skins or themes, different firmware versions) but it only shows up when the auto bed leveling is being turned on. 

I guess, it's time for Makerbase to have a look at the code and check the storage routine of the auto bed leveling.

technik-gegg, 2018-10-25:
Ok, looks like I found a solution to that...

I figured, that some of my user defined buttons also changes the fore- and background colors. So I've dug a little deeper and found this:
If you create user defined buttons, you apply GCODE to those buttons and, as stated somewhere in the config file, you have to take care to end each line with a semicolon (;). 
What this hint doesn't say is: Right after the last semicolon, there has to be a CR/LF - no spaces, no tabs, no what so ever.
I've checked my config and found two of them having spaces afterwards. I removed the spaces and.. bingo... no more color changes.

Now: If  you watch the auto bed leveling command carefully, you'll notice the "#the setting of auto-leveling command button" comment at the end of the line.
Simply move to the end of the GCODE sequence (G28;G32 S2;) and key in a CR and your are problems gone.

Conclusion: As I stated in my first post, this might be a flaw in the firmware, overwriting some internal memory.
However, this work around fixes it.

vk220, 2018-10-28:
> Ok, looks like I found a solution to that...

Well, You're cool. It works.
I didn't create user-defined buttons, so didn't dig so deep.
Thank you!

vk220, 2018-10-29:
> The Z-Probe measurement

May I ask which sensor are you use?

technik-gegg, 2018-10-30:
The very standard LJ12A3-4-Z/BX.

Be aware, that this type of Z-Probe can't be wired directly to the board, since it runs on 12/24V and the output signal would fry your controller. Instead, I'm using a transistor in between the output signal and the endstop input (as shown in this very sophisticated schematic: http://forum.arduino.cc/index.php?topic=425090.msg2929575#msg2929575).
Any NPN transistor will do the job, i.e. BC547. The resistor show in the picture is about 100K.

Also, my Z-Probe goes to  Z-min, whereas the Z-Axis homes to Z-max. This is the only way to prevent the Nozzle crashing into the bed occasionally.

However, like many of the users around here, I've tried to find some decent use for the Robin I got lying around almost a year. Even though I got it up and running now, I will dump the board and move to the Duet 2  Maestro instead. 

I've build 2 printers so far - one of them is a multi tool printer with 4 Extruders - using the Duet 2 (WiFi and Ethernet versions) and I'm totally spoiled.
Not because it's Open Source, it's because the huge amount of information you get from the vendor itself as well as from the forums. Not a single G-Code is left undocumented.

Sure, it's double the price of the Robin but from my point it's totally worth it. Once you've tasted the sweetness to control and configure your printers from within any browser, without the need of a touchscreen display, it's hard to go back to anything else.
Updates come regularly and bring you bug fixes and new features - this goes way beyond the "we delete some files and upload it again to keep the crowd thinking we're working on the software"-politics which I can see in here.

However, trying to get something to work even if it seems pointless, is a kind of a challenge and adventure. In this sense, good luck to you and your project :)

technik-gegg, 2018-10-30:
Btw.

Z-Probe **measurement** is kind of misleading. This is more like triggering, just as a normal switch would do. For a real measurement you'd need a connection of the Z-Probe output to a analog input on the controller, which is not the case, as far as I know.

3Dangel, 2020-01-04:
just putting here that the new MArlin 2 is now fully dupported on this MKS Robin