makerbase-mks/MKS-Robin #65: 2 extruder motors with one hot end

mattdog01, 2018-10-27:
Has anyone had success with this configuration?
I am using (going to use) a Y-adapter to alternate between two different filaments to the same hot end.  Is there a way to specify that there is only one hotend so that I don't need to have a dummy hotend sitting there with a heater and temp sensor?
I thought this had come up in a post but I don't see it.
Thanks for your help.

innc1, 2019-09-29:
I connected the thermistor cable to both heater pins.the second nozzle sees it as ambient temperature when it starts to heat up. But not tried using second nozzle yet .

mattdog01, 2020-01-09:
My final configuration was to connect both heater outputs together, though diodes, to one heater cartridge in the hot end. I drilled the hole in the heater block deeper so I could put two thermistors in there.  The printing works great and temp control was excellent.  
Now we are in a post Marlin 2.0 era and there is a no long a config issue.  The configuration is build into the firmware and now my second thermistor is a safety backup to the main thermistor.  If their temp readings get too far apart the heater is shut down. No need to connect both outputs together through the diodes.  Life is good.  Thanks Marlin team.

closed 2020-01-09 by mattdog01