makerbase-mks/MKS-Robin #66: sd card limits

mattdog01, 2018-10-27:
Is there a limit to the size, type or format of an SD card used on the MKS-Robin?
I have a new 64gig sdxc card and the Robin can't see any of the files.  :(
I have a 2nd card, smaller and straight SD and it works just fine.
Thanks for any help you can give.

valentinoo0, 2018-10-31:
Now im using 32gb and works good but I dont know if could support 64 gb sorry mate.

mattdog01, 2018-11-01:
Thanks for the info valentinoo0.  I guess 64gb is pushing the limits.  All the best.

mattdog01, 2018-11-26:
After a bit of research I found out that the Robin needs FAT32 file partitions and that Windows with not partition in FAT32 above 32Gig SD cards.  I found a utility that for now is free to use and it worked perfectly.  I reformatted my 64 Gig SD card in FAT32 and it works like a charm.  The utility is called "AOMEI Partition Assistant Standard Edition 7.5.1".  SO if you want to use a large SD card with your Robin this works.  Happy Printing.

closed 2018-11-26 by mattdog01