makerbase-mks/MKS-Robin #67: z-probes compatibility

vk220, 2018-10-29:
Hi! That's not a secret that many people waiting for BLTouch support (also Iam bought this). But it seems like we'll be waiting for a long time. For some hidden reasons, maybe, that will never be. In additional, we have very poor documentation, at last in English.
Nevertheless, I've create this issue, where I want to ask to share which sensors are compatible with MKS robin?
I used [this sensor](https://ru.aliexpress.com/item/1pc-3D-Printer-Parts-Auto-Leveling-Sensor-Module-For-Makerbot-Ultimaker-MK8-E3D-E3D-V6-Nozzle/32795009159.html?spm=a2g0s.9042311.0.0.274233edlHnYw8) and it really works, but the very inconveniense is that you have always remove it before nozzle heating.
I want try [this](https://ru.aliexpress.com/item/3D-Printer-P-I-N-D-A-Probe-Leveling-Sensor-Compatiable-With-mini-rambo-For-DIY/32908456490.html?spm=a2g0v.10010108.1000016%2FB.1.22ff66327HOQFQ&isOrigTitle=true)
There is schematic on the store shots:
[image]
and I guess it will work. But maybe there are better solutions?
Well, thanks.

makerbase-mks, 2018-10-31:
Hello, because the robin board is a closed source motherboard, there is no bltouch interface reserved, so it is temporarily not supported. So generally the leveling sensors are all connected to Zmin. Because of the variety of sensors, there is no way to test each. But one thing to note is that the s terminal of Zmin on the robin motherboard can only accept 3.3v voltage, so the sensor return signal can't be higher than 3.3v.

vk220, 2018-10-31:
> But one thing to note is that the s terminal of Zmin on the robin motherboard can only accept 3.3v voltage, so the sensor return signal can't be higher than 3.3v.

Hi, thank you for answer!
In my opinion - it's quite useful information. 
And although robin board is closed-source, there will be good to have specs for other ports.
Anyway, thanks.