makerbase-mks/MKS-Robin #68: engines do not respond to commands

YuraBz, 2018-11-06:
Hi, I connected the driver drv8825 to the board, started running with the default configuration and driver once trying to scroll and no longer responds to the commands. checked various drivers and engines, the result is the same. In the application on the screen hang the inscription "error"

YuraBz, 2018-11-06:
checked different vref, acceleration, speed, step, the result is the same. maybe someone has a working configuration file under xyz

makerbase-mks, 2018-11-19:
The normal condition of the Robin board to work is to be able to check the temperature from the thermal port.

YuraBz, 2018-11-21:
good day, With this I have taken care of. can one more question? 
І have mks slot 1.1 8pin and mks robin 2.4 with 9pin sd external port, can i connect them somehow?

makerbase-mks, 2018-11-22:
No,the definition of pins are difference.You can replace it with mks slot2.

YuraBz, 2018-11-22:
where can i find pinout or datasheet slot 1.1 & 2?