makerbase-mks/MKS-Robin #70: mks robin V2.4 mainboard update problem

emreyz, 2018-11-14:
After ı try to config and update, my robin V2.4 board,  screen turn to white. and ı could not do anything.
could any body help me solved this issue.

what is the suitable firmware with robin 2.4 board

emreyz, 2018-11-14:
I try 2.03 and 2.06 versions 2.06 version do on the screen "complete" but ı never do anything , after that ı ty 2.03 and turn white screen again same ı could not make anything

GrafZahl81, 2018-11-21:
My MKS Robin 2.4 with TFT V2.0 does not update the mks_pics folder.
 It always gets stuck with bmp_Add.bin. It does not start after that.

makerbase-mks, 2018-11-22:
@GrafZahl81  you can try to replace the sd and update again.
@emreyz If you receive a broken motherboard, you can contact our customer service to handle it.

pk3pdk, 2019-01-18:
@emreyz I had the same problem with the white screen after first start up, after using the MKS_Robin_V2.0.7 Release FIle screen works after SD Card update.
I´m using Robin V2.4 and TFT V2.0 from August 2018...
I did not connected anything except 12V Power and TFT. But I hope it works.

emreyz, 2019-01-20:
we can use just V2.0.7 to suitable v2.4 best regards

________________________________
Gönderen: pk3pdk <notifications@github.com>
Gönderildi: 18 Ocak 2019 Cuma 20:32
Kime: makerbase-mks/MKS-Robin
Bilgi: emreyz; Mention
Konu: Re: [makerbase-mks/MKS-Robin] mks robin V2.4 mainboard update problem (#70)

@emreyz<https://github.com/emreyz> I had the same problem with the white screen after first start up, after using the MKS_Robin_V2.0.7 Release FIle screen works after SD Card update.
I´m using Robin V2.4 and TFT V2.0 from August 2018...
I did not connected anything except 12V Power and TFT. But I hope it works.

adarshparavoor, 2019-03-10:
I got the MKs V2.4 board i don't know how to configure this board. Any link with essential details

makerbase-mks, 2019-03-11:
@adarshparavoor
you can get the datasheet  on this  web
https://github.com/makerbase-mks/Datasheet/tree/master/English%20datasheet
If you need firmware, you can download it through this website.

https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin

adarshparavoor, 2019-03-13:
> cfg_background_color			0x000000		#the color of screen background
> cfg_title_color			0xFFFFFF		#the color of title text
> cfg_state_background_color		0x000000		#the color of temperature,fan,(etc)background
> cfg_state_text_color			0xFFFFFF		#the color of temprature,fan,(etc) text
> cfg_filename_background_color		0x000000		#the color of file button
> cfg_filename_color			0xFFFFFF		#the color of file text
> cfg_printingstat_word_background_color	0x000000		#the color of background in printing status
> cfg_printingstat_word_color		0xFFFFFF		#the color of text in printing status
> cfg_BUTTON_3D_effects		0				#3D effects setting (disable: 0, enable: 1)

sir,

My printer  is a custom made core xy printer. board that i using MKS Robin
V 2.0 End stop using here is optical sensor. The code that i used for the
purpose is given below.

* X axis End stop is not working when the homing direction is as on the
code given below. If i reverse this direction end stop will work. Please
help me to solve this issue.

*Is it possible to add capacitive sensor then how can i add the sensor.

I need more details to finish the work. Please help me fast

> @adarshparavoor <https://github.com/adarshparavoor>
> you can get the datasheet on this web
> https://github.com/makerbase-mks/Datasheet/tree/master/English%20datasheet
> If you need firmware, you can download it through this website.
>
> https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin

#-------------------------------------------------------------
############ Based Settings( Required) #######################
#machine type
#0:xyz
#1:corexy(x_motor = x+y, y_motor = x-y),
#2:corexy(x_motor = x+y, y_motor = y-x),
#3:delta

>DRIVE_SYSTEM 		        2				#machine type

>cfg_multiple_language		1				#multi-language(enable:1, disable:0)

>cfg_language_type		3				#languages setting,this configuration is valid when "cfg_multiple_language" is disabled. 
								#(simplified Chinese:1; traditional Chinese:2; English:3; Russian:4; Spanish:5;French:6;Italian:7).

>NUM_EXTRUDER 			1				#number of extruder (1:singal；2:dual)
>HAVE_HEATED_BED 		1				#1:enable bed;  0:disable bed
>EXT0_TEMPSENSOR_TYPE 		1				#1:100k thermistor；102: MAX31855 thermocouple
>EXT1_TEMPSENSOR_TYPE 		1				#1:100k thermistor；102: MAX31855 thermocouple
>HEATED_BED_SENSOR_TYPE 	1				#1:100k thermistor；

>FEATURE_TWO_XSTEPPER		0				#change E1 singal into X , then dual X is available(enable:1; disable: 0)
>FEATURE_TWO_YSTEPPER		0				#change E1 singal into Y , then dual Y is available(enable:1; disable: 0)
>FEATURE_TWO_ZSTEPPER		0				#change E1 singal into Z , then dual Z is available(enable:1; disable: 0)

#--------------------------------------------------------------
################# Machine settings ############################
>EXT1_X_OFFSET			0				#E1 offset from the origin of X axis (mm)
>EXT1_Y_OFFSET			0				#E1 offset from the origin of Y axis (mm)

>INVERT_X_DIR			0				#X motor direction, 1 goes opposite direction;
>INVERT_Y_DIR			0				#Y motor direction, 1 goes opposite direction;	
>INVERT_Z_DIR			0				#Z motor direction, 1 goes opposite direction;	
>EXT0_INVERSE 			1				#E0motor direction, 1 goes opposite direction;
>EXT1_INVERSE 			0				#E1motor direction, 1 goes opposite direction;

>XAXIS_STEPS_PER_MM		100				#X steps per mm 
>YAXIS_STEPS_PER_MM		100				#Y steps per mm 
>ZAXIS_STEPS_PER_MM		400				#Z steps per mm 
>EXT0_STEPS_PER_MM		100				#E0 steps per mm 
>EXT1_STEPS_PER_MM		100				#E1 steps per mm 

>X_MAX_LENGTH 			370				#the MAX X-axis distance
>Y_MAX_LENGTH 			410				#the MAX Y-axis distance
>Z_MAX_LENGTH 			400				#the MAX Z-axis distance
>X_MIN_POS 			0				#the MIN X-axis distance
>Y_MIN_POS 			0				#the MIN Y-axis distance
>Z_MIN_POS 			0				#the MIN Z-axis distance

[... 189 lines cut ...]
>moreitem_button2_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button3_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button4_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button5_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button6_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button7_cmd:G28 X0;G28 Y0;G28 Z0;

#the number of "more" button to dispaly on printing interface
>morefunc_cnt:0

#button1~button6 function 
>morefunc1_cmd:G28;
>morefunc2_cmd:G28;
>morefunc3_cmd:G28;
>morefunc4_cmd:G28;
>morefunc5_cmd:G28;
>morefunc6_cmd:G28;

#----------------------------------------------------------------
##############  Display Customization  #######################

makerbase-mks, 2019-03-14:
HI
The direction you set back to zero is this
X_HOME_DIR 1 #Homing direction(-1:MIN,1:MAX)
Y_HOME_DIR -1 #Homing direction(-1:MIN,1:MAX)
Z_HOME_DIR 1 #Homing direction(-1:MIN,1:MAX)
That is, x and z will reverse max, y back to min direction

But you enable the hardware limit but enable xyz to be the minimum limit.
# 1 for Min/Max endstop enable in hardware, while 0 disable
MIN_HARDWARE_ENDSTOP_X 1
MIN_HARDWARE_ENDSTOP_Y 1
MIN_HARDWARE_ENDSTOP_Z 1
MAX_HARDWARE_ENDSTOP_X 0
MAX_HARDWARE_ENDSTOP_Y 0
MAX_HARDWARE_ENDSTOP_Z 0

Therefore, it is said that the limit does not work. For example, if you set x to max, but you are exposed to the xmin limit switch, or a max switch that is not enabled, it is not normal.

If you want to press your direction of zero return, the limit you enable should be like this
# 1 for Min/Max endstop enable in hardware, while 0 disable
MIN_HARDWARE_ENDSTOP_X 0
MIN_HARDWARE_ENDSTOP_Y 1
MIN_HARDWARE_ENDSTOP_Z 0
MAX_HARDWARE_ENDSTOP_X 1
MAX_HARDWARE_ENDSTOP_Y 0
MAX_HARDWARE_ENDSTOP_Z 1

If wiring, you should also connect to the corresponding interface of min or max on the motherboard.

maocastillo, 2019-04-16:
Hi. I am developing a large format printer and I use the MKS ROBIN v2.4 board. I do the configuration of the board (firmware) and it works very well, but I turn off the machine and turn it on again and the configuration is lost and returns to the factory firmware. I have changed several SD cards of various types and the same thing always happens, there is no firmware left on the board, the same thing happens to it. I appreciate your help thanks

github051, 2019-09-07:
How to change the jerk settings ?

Drakstein, 2019-11-20:
I have a simple question about the voltage of the MKS ROBIN v2.4 board, it suports input 24V, but it also outputs 24V?? By the way, I will use a BL touch, anyone here already used it, any suggestions? I will use in a core XY model.