makerbase-mks/MKS-Robin #71: steps/mm

killerage, 2018-12-12:
even if I change values the movement doesn't change , any idea?

killerage, 2018-12-12:
so I managed to move it , but I have to got to 800 which makes the printer super fast I can't print like that

mattdog01, 2018-12-13:
Hello Killerage,
I'm not sure what the issue is but I'll try to help.  
To change the configuration of the Robin you need to change the config file when it is in TEXT FILE (.txt) format and then save that to your sd card, put it in the Robin and then turn on power to start the Robin.  When the Robin starts it reads the .txt file makes the changes to the firmware settings and then it edits the file extension type so that it does read it again.  If you want to changes the settings again then the config file needs to be in the .txt format or at startup the Robin ignores the file.
To set your steps per mm use the Prusa calculator for x, y and z.  
Find that here>>>https://www.prusaprinters.org/calculator/
E is calibrated by running some filament through the extruder and measuring how much went through.
I hope this helps but if not repost and we can try again.
Matt