makerbase-mks/MKS-Robin #72: help  do not recognize the dimensions of my printer

nilose2014, 2018-12-28:
Hello
I have a problem and it is that I do not recognize the dimensions of my printe
[robin_config.txt](https://github.com/makerbase-mks/MKS-Robin/files/2713664/robin_config.txt)
r and I think I have it properly configured is an e12 anet whose dimensions are 300x300x400, someone can help me upload the config.txt

>X_MAX_LENGTH 			300				#the MAX X-axis distance
>Y_MAX_LENGTH 			300				#the MAX Y-axis distance
>Z_MAX_LENGTH 			400				#the MAX Z-axis distance
>X_MIN_POS 			-5				#the MIN X-axis distance
>Y_MIN_POS 			-26				#the MIN Y-axis distance
>Z_MIN_POS 			0				#the MIN Z-axis distance

makerbase-mks, 2018-12-29:
what is type of your machine?
The XYZ machine need to set:
 X_MIN_POS 0	#the MIN X-axis distance
Y_MIN_POS 0	#the MIN Y-axis distance
Z_MIN_POS 0	#the MIN Z-axis distance
The delta machine need to change below settings:
################ Delta Settings ##############################
>DELTA_MAX_RADIUS				135		#the radius of Delta annulus
>PRINTER_RADIUS					197		#the distance from machine center to vertical top
>DELTA_DIAGONAL_ROD				346.75		#the length of Delta pole
>DELTA_FLOOR_SAFETY_MARGIN_MM			15		#the safe distance of leveling edge
>END_EFFECTOR_HORIZONTAL_OFFSET			28.0		#
>CARRIAGE_HORIZONTAL_OFFSET 			14.5		#
>ROD_RADIUS		 			169		#
You can

nilose2014, 2018-12-29:
Hello
Muy machine is XYZ model Anet e12

nilose2014, 2018-12-29:
Hello

I have it on

X_MIN_POS  -25 #the MIN X-axis distance
Y_MIN_POS -6 #the MIN Y-axis distance
Z_MIN_POS 0 #