makerbase-mks/MKS-Robin #73: Possibility to invert Enable for stepper

0Jmaster0, 2019-01-08:
Hello
I bought 1/256 stepper controllers ST820 which are pin to pin compatible with conrtollers like a4988, drv8825 etc. But they have one difference - their enable pin is reversed in comparison to a4988. Can you add possibility to change behavior of enable pins to be reversed? 

Thank you for answer

3Dangel, 2019-01-08:
> Forget to add - my board is MKS Robin 2.4

Do the hardware jump
Will be quicker
Waiting for reply will frustrate you.

Cheers

makerbase-mks, 2019-01-09:
In the firmware of 2.0.7, there is this configuration item in the configuration file.

#For Inverting Stepper Enable Pins
>X_ENABLE_ON		0
>Y_ENABLE_ON		0
>Z_ENABLE_ON		0
>EXT0_ENABLE_ON		0
>EXT1_ENABLE_ON		0

0Jmaster0, 2019-01-09:
Thank you! Didn't notice that in config :)

closed 2019-01-09 by 0Jmaster0