makerbase-mks/MKS-Robin #75: What is the Robin2?

mattdog01, 2019-01-25:
Is the Robin2 a new controller board?  I don't seem to be able to find any info about it on the Makerbase-MKS site.  Looking at the config file it has support for the BLTouch and some other stuff.  Does anyone know about this?

mattdog01, 2019-01-28:
Hello 3Dangel,
The question is about the Robin2 and what info there is out there.  It doesn't seem you have looked at the config file for Robin2 as it specifically mentions the BL-touch.  My hope is that the Robin2 is not a different board and is the old Robin with a brand new alternate firmware which would be wonderful but not likely.
Also Github is to be used for each issue separately please don't combine comments for different issues. With my other post I'm trying to find out if someone with a Robin has had issues with the firmware calculations.  I also have posts in other places.  Thanks for your thoughts.
Take care and happy printing.

Here a section of ROBIN2_cfg
#============================= Z Probe Options =============================
>BLTOUCH			0			# 0:disable BLTOUCH; 1:enable BLTOUCH
#Select for a probe connected to Z-Min or Z-Max.
>Z_MIN_PROBE_PIN_MODE		0	# 0 : NULL; 1: ZMIN; 2: ZMAX

>Z_PROBE_OFFSET_FROM_EXTRUDER	0	# Z offset: -below +above  [the nozzle]		
>X_PROBE_OFFSET_FROM_EXTRUDER	0	# X offset: -left  +right  [of the nozzle]
>Y_PROBE_OFFSET_FROM_EXTRUDER	0	# Y offset: -front +behind [the nozzle]
>XY_PROBE_SPEED 		4000	# X and Y axis travel speed (mm/m) between probes
>Z_PROBE_SPEED_FAST 		600	# Speed for the first approach when double-probing (with PROBE_DOUBLE_TOUCH)
>Z_PROBE_SPEED_SLOW 		300	# Speed for the "accurate" probe of each point

3Dangel, 2019-01-28:
True
Did not see that.
It was alway said that it was not supported.
I guess, I need to update my own info

Cheers

vk220, 2019-03-08:
Also interested.
In order of experiment I tried to flash this to my mks robin v 2.3. The only way I found was renaming Robin2.bin to Robin.bin. It flashed successfully but it doesn't seem to work. It just says "Complete" right after booting.
So I reverted to 2.0.7 firmware.

makerbase-mks, 2019-03-14:
The firmware of robin and robin2 is not universal, and robin2 supports bltouch. The robin firmware is not supported, and the corresponding socket is not reserved.

mattdog01, 2019-03-14:
Thank you Makerbase-mks. The issue can now be closed.

closed 2019-03-14 by mattdog01