makerbase-mks/MKS-Robin #78: Hot End home position Help

TCWC, 2019-03-01:
I have a X3A and just installed the MKS Robin 2.4 w/ V2.0.7 firmware. I figured out how to change and update the config.txt file but i'm having a hard time figuring out what parameter in the config.txt file controls where the hot end stops after a all home cmd. X and Y goes to the end stops and stay. Then the Z goes down but not over the bed so it bottoms out.

TCWC, 2019-03-02:
#-------------------------------------------------------------
############ Based Settings( Required) #######################
#machine type
#0:xyz
#1:corexy(x_motor = x+y, y_motor = x-y),
#2:corexy(x_motor = x+y, y_motor = y-x),
#3:delta

>DRIVE_SYSTEM 		        0				#machine type

>cfg_multiple_language		1				#multi-language(enable:1, disable:0)

>cfg_language_type		3				#languages setting,this configuration is valid when "cfg_multiple_language" is disabled. 
								#(simplified Chinese:1; traditional Chinese:2; English:3; Russian:4; Spanish:5;French:6;Italian:7).

>NUM_EXTRUDER 			1				#number of extruder (1:singal£»2:dual)
>HAVE_HEATED_BED 		1				#1:enable bed;  0:disable bed
>EXT0_TEMPSENSOR_TYPE 		1				#1:100k thermistor£»102: MAX31855 thermocouple
>EXT1_TEMPSENSOR_TYPE 		1				#1:100k thermistor£»102: MAX31855 thermocouple
>HEATED_BED_SENSOR_TYPE 	1				#1:100k thermistor£»

>FEATURE_TWO_XSTEPPER		0				#change E1 singal into X , then dual X is available(enable:1; disable: 0)
>FEATURE_TWO_YSTEPPER		0				#change E1 singal into Y , then dual Y is available(enable:1; disable: 0)
>FEATURE_TWO_ZSTEPPER		0				#change E1 singal into Z , then dual Z is available(enable:1; disable: 0)

#---------------------------------------------------------------
################# Machine settings ############################
>EXT1_X_OFFSET			0				#E1 offset from the origin of X axis (mm)
>EXT1_Y_OFFSET			0				#E1 offset from the origin of Y axis (mm)

>INVERT_X_DIR			1				#X motor direction, 1 goes opposite direction;
>INVERT_Y_DIR			1				#Y motor direction, 1 goes opposite direction;	
>INVERT_Z_DIR			0				#Z motor direction, 1 goes opposite direction;	
>EXT0_INVERSE 			1				#E0motor direction, 1 goes opposite direction;
>EXT1_INVERSE 			0				#E1motor direction, 1 goes opposite direction;

>XAXIS_STEPS_PER_MM		100				#X steps per mm 
>YAXIS_STEPS_PER_MM		100				#Y steps per mm 
>ZAXIS_STEPS_PER_MM		400				#Z steps per mm 
>EXT0_STEPS_PER_MM		100				#E0 steps per mm 
>EXT1_STEPS_PER_MM		100				#E1 steps per mm 

>X_MAX_LENGTH 			220				#the MAX X-axis distance
>Y_MAX_LENGTH 			220				#the MAX Y-axis distance
>Z_MAX_LENGTH 			300				#the MAX Z-axis distance
>X_MIN_POS 			0				#the MIN X-axis distance
>Y_MIN_POS 			0				#the MIN Y-axis distance
>Z_MIN_POS 			0				#the MIN Z-axis distance

>MIN_EXTRUDER_TEMP		175				#MIN TEMP on Extruder ,play a protective role
>MAX_EXTRUDER_TEMP 		275				#MAX TEMP on Extruder ,play a protective role
>MAX_HEATED_BED_TEMP 		150				#MAX TEMP on heated bed £¬play a protective role

>HOMING_ORDER			1				#Set direction of endstops when homing;      
>X_HOME_DIR			-1				#Homing direction(-1:MIN£¬1:MAX)
>Y_HOME_DIR			-1				#Homing direction(-1:MIN£¬1:MAX)
>Z_HOME_DIR			-1				#Homing direction(-1:MIN£¬1:MAX)

>HOMING_FEEDRATE_X 		40				#the feedrate on X homing
>HOMING_FEEDRATE_Y 		40				#the feedrate on Y homing
>HOMING_FEEDRATE_Z 		10				#the feedrate on Z homing

# 1 means endstop always-on,0 is always-off
>ENDSTOP_X_MIN_INVERTING 	1				
>ENDSTOP_Y_MIN_INVERTING 	1				
>ENDSTOP_Z_MIN_INVERTING 	1				
>ENDSTOP_X_MAX_INVERTING 	0				
>ENDSTOP_Y_MAX_INVERTING 	0				
>ENDSTOP_Z_MAX_INVERTING 	0				

[... 186 lines cut ...]
>morefunc3_cmd:G28;
>morefunc4_cmd:G28;
>morefunc5_cmd:G28;
>morefunc6_cmd:G28;

#----------------------------------------------------------------
##############  Display Customization  #######################

>cfg_background_color			0x000000		#the color of screen background

>cfg_title_color			0xFFFFFF		#the color of title text

>cfg_state_background_color		0x000000		#the color of temperature,fan,(etc)background
            
>cfg_state_text_color			0xFFFFFF		#the color of temprature,fan,(etc) text

>cfg_filename_background_color		0x000000		#the color of file button

>cfg_filename_color			0xFFFFFF		#the color of file text

>cfg_printingstat_word_background_color	0x000000		#the color of background in printing status

>cfg_printingstat_word_color		0xFFFFFF		#the color of text in printing status

>cfg_BUTTON_3D_effects		0				#3D effects setting (disable: 0, enable: 1)

TCWC, 2019-03-13:
Not sure whats going on with you guys (MKS). But i  bought your product and was looking for help. 11days now and no response. But you respond to other post within a couple days. So i guess i need to look for a new product. Thanks

3Dangel, 2019-03-13:
Manually lift your z up high
Then home again
And manuallybtrigger your z end stop by hand
See if it stops your z
If not
You have installed wrong your endstop

Cheers

TCWC, 2019-03-13:
Thanks for the response. End stop is right and i get it to stop now. Just having issues with i guess to be build area and offsets. I can control this in S3D but in the config.txt it doesnt seem to help.

3Dangel, 2019-03-13:
with anything that is 3d printing and on the cheap side, it will require a
LOT of tinkering and try on errors.
join reprap forum and seek more help that way....
MKS makes good quality boards, but I think their are mainly focused in
mainland China's market share....
wish you luck

TCWC, 2019-03-13:
Thanks for the info. I do know that they need a lot of tinkering. Been doing that since day one. I guess I just was hoping that companies that make good boards for cheap equipment and send them here lend a little help when needed. Especially when its closed source. I like the board. never said I didn't. But again thanks for the info Ill try RepRap.