makerbase-mks/MKS-Robin #79: MKS ROBIN V 2.4 DOUBLE MOTOR PROBLEM Z DOES NOT WORK

roberto76milano, 2019-03-06:
hello I own the version 2.4 robin card I did upgrade firmware but I can not run the second engine of z, I selected option in the .txt file but it does not work !!!
I installed version 2.7, can you help me ????
over all the engines move only 5mm instead of 10mm, I use purple driver 8825 how should I set ???

roberto76milano, 2019-03-06:
hello I own the version 2.4 robin card I did upgrade firmware but I can not run the second engine of z, I selected option in the .txt file but it does not work !!!
I installed version 2.7, can you help me ????
over all the engines move only 5mm instead of 10mm, I use purple driver 8825 how should I set ???

roberto76milano, 2019-03-06:
my printer is a prusa geeetech i3 mk8

vk220, 2019-03-08:
Hi! How did you connect 2nd z-motor? Into the E1 port?
And did you set amount XAXIS_STEPS_PER_MM and others in the .txt file?

makerbase-mks, 2019-03-14:
Do you have this item set in the configuration file?
Just change E1 to the second z-axis, and set it up here.
Also note that the E1 axis is also to be driven.

>FEATURE_TWO_XSTEPPER		0				#change E1 singal into X , then dual X is available(enable:1; disable: 0)
>FEATURE_TWO_YSTEPPER		0				#change E1 singal into Y , then dual Y is available(enable:1; disable: 0)
>FEATURE_TWO_ZSTEPPER		1				#change E1 singal into Z , then dual Z is available(enable:1; disable: 0)