makerbase-mks/MKS-Robin #80: X and Y offset for z-probe

vk220, 2019-03-08:
Hi! As far as I know for Z offset of the z-probe I should use the Z_PROBE_HEIGHT parameter.
But I wonder if there are offsets for x and y axes?
In other topics I saw something like:
> Z_PROBE_X_OFFSET xx
> Z_PROBE_Y_OFFSET xx
and also in another topic:
X_PROBE_OFFSET_FROM_EXTRUDER
Y_PROBE_OFFSET_FROM_EXTRUDER
Well, will some of them work?
Thanks!

vk220, 2019-03-09:
Okay, tried:
```
>Z_PROBE_X_OFFSET xx
>Z_PROBE_Y_OFFSET xx
```
And it seems to be work.

makerbase-mks, 2019-03-14:
Yes, this is to adjust the automatic leveling, similar to the z-probe leveling offset value?

makerbase-mks, 2019-03-14:
Yes, this is to adjust the auto-leveling, similar to the z-probe leveling offset values

closed 2019-03-14 by makerbase-mks

hinotori74, 2020-02-21:
Sorry I do not speak English. All this is wrong. The correct parameter for MKS Robin is:

>Z_PROBE_OFFSET_X nn
>Z_PROBE_OFFSET_Y nn

Another possibility is from pronterface run:

M206 T3 P800 Xnn
 
I use this script for Auto Leveling:

>cfg_auto_leveling_cmd:G28 XY;G1 X110 Y50;G28 Z;G1 Z5;G32 S2;

My >Z_PROBE_OFFSET_X is 40mm so this is esential: ;G1 X110 Y50; from line above. if not the x axis hit with the end-stop.

hinotori74, 2020-02-21:
sorry, the only thing that works is to run from pronterface M206 ..... My hardware is MKS Robin 2.4 whit firmware 2.08

makerbase-mks, 2020-02-21:
>Z_PROBE_HEIGHT				-0.8			#the height difference between Z-probe and  nozzle
This configuration value is the value of Z-offset. Robin does not open the configuration items of X and Y offset values.

hinotori74, 2020-02-21:
But with the M206 command, work great. Example M206 T3 P800 X40. if you set 50.000 mm on Z-probe X1 and 50.000 mm on Z-probe Y1, when execute Autolevel the first point is X10 (+40 of offset =50) Y50. The probe is located at the exact point.

hinotori74, 2020-02-21:
This is my partial configuration:

>cfg_auto_leveling_cmd:G28 XY;G1 X110 Y50;G28 Z;G1 Z5;G32 S2;						#the setting of auto-leveling command button

>FEATURE_Z_PROBE								  1		#0:disable leveling£¬1:enable leveling
>BED_LEVELING_METHOD							  1		#0:3 points leveling,1:more points leveling,2: 4 points leveling
>Z_PROBE_ON_HIGH								  1		#Probe signal(0: low level,always on; 1:high level,always off)

>Z_PROBE_OFFSET_X 								 40.0	#NOT WORK - **Use M206 T3 P800 X40.00** 
>Z_PROBE_OFFSET_Y 								  0		#NOT WORK - Use M206.........
>Z_PROBE_HEIGHT									  0.5 	#the height difference between Z-probe and  nozzle

>Z_PROBE_X1										 50		#coordinateX1 is preset point
>Z_PROBE_Y1										 50		#coordinateY1 is preset point

>Z_PROBE_X2										170		#coordinateX2 is preset point
>Z_PROBE_Y2										 50		#coordinateY2 is preset point

>Z_PROBE_X3									     50		#coordinateX3 is preset point
>Z_PROBE_Y3										205		#coordinateY3 is preset point