makerbase-mks/MKS-Robin #81: MKS Robin2

JanHH2019, 2019-03-10:
After a long search, I have now got the MKS Robin2 board and would like to operate it with a BLTouch. Meanwhile, I found out how to connect it and initalization works well.
Unfortunately, I find nowhere a guide to that, what is in the configuration file does not really help me.
How do I use the BLTouch now?

makerbase-mks, 2019-03-11:
1. Set the leveling mode
#=========================== Leveling button settings  =============================

>cfg_leveling_mode			1			#1:auto-leveling; 0:manual leveling

2. Enable the corresponding limit, set the limit switch mode of z-probe (delta, generally bltouch is connected to z-, because z+ has been used. The i3 structure is reversed, generally it is connected to z+. Which one is used below is Have settings)The normally open and normally closed type of the endstops, set according to your own situation, do not completely copy my configuration
# Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
>X_MIN_ENDSTOP_INVERTING 0		# set to true to invert the logic of the endstop.
>Y_MIN_ENDSTOP_INVERTING 0		# set to true to invert the logic of the endstop.
>Z_MIN_ENDSTOP_INVERTING 1		# set to true to invert the logic of the endstop.
>X_MAX_ENDSTOP_INVERTING 1		# set to true to invert the logic of the endstop.
>Y_MAX_ENDSTOP_INVERTING 1		# set to true to invert the logic of the endstop.
>Z_MAX_ENDSTOP_INVERTING 1		# set to true to invert the logic of the endstop.
>Z_MIN_PROBE_ENDSTOP_INVERTING 1	# set to true to invert the logic of the probe.
>FIL_RUNOUT_INVERTING 	0			# set to true to invert the logic of the Filament Runout Sensor.

# Specify here all the endstop connectors that are connected to any endstop or probe.
>USE_XMIN_PLUG	1			# 1:used; 0:noused
>USE_YMIN_PLUG	1			# 1:used; 0:noused
>USE_ZMIN_PLUG	1			# 1:used; 0:noused
>USE_XMAX_PLUG	1			# 1:used; 0:noused
>USE_YMAX_PLUG	0			# 1:used; 0:noused
>USE_ZMAX_PLUG	0			# 1:used; 0:noused

3. Enable bltouch, and select the port to be connected (z- or z+，set according to your own situation)
#============================= Z Probe Options =============================
>BLTOUCH			1			# 0:disable BLTOUCH; 1:enable BLTOUCH
#Select for a probe connected to Z-Min or Z-Max.
>Z_MIN_PROBE_PIN_MODE		2	# 0 : NULL; 1: ZMIN; 2: ZMAX

>Z_PROBE_OFFSET_FROM_EXTRUDER	0	# Z offset: -below +above  [the nozzle]		
>X_PROBE_OFFSET_FROM_EXTRUDER	0	# X offset: -left  +right  [of the nozzle]
>Y_PROBE_OFFSET_FROM_EXTRUDER	0	# Y offset: -front +behind [the nozzle]
>XY_PROBE_SPEED 		4000	# X and Y axis travel speed (mm/m) between probes
>Z_PROBE_SPEED_FAST 		600	# Speed for the first approach when double-probing (with PROBE_DOUBLE_TOUCH)
>Z_PROBE_SPEED_SLOW 		300	# Speed for the "accurate" probe of each point

4. The range and points of the final leveling（The level of the leveling does not exceed the range of the trip）
#=============================== Bed Leveling ==============================

>BED_LEVELING_METHOD 		3	# 0:NULL_BED_LEVELING; 3:AUTO_BED_LEVELING_BILINEAR; 5:MESH_BED_LEVELING	

>GRID_MAX_POINTS_X 		3	# the number of grid points per dimension. <= 15
>GRID_MAX_POINTS_Y 		3	# the number of grid points per dimension. <= 15	
>Z_CLEARANCE_DEPLOY_PROBE	20	# Z Clearance for Deploy/Stow	   > 0
>Z_CLEARANCE_BETWEEN_PROBES	20	# Z Clearance between probe points  > 0

# Set the boundaries for probing (where the probe can reach).
>LEFT_PROBE_BED_POSITION 	30	
>RIGHT_PROBE_BED_POSITION 	180
>FRONT_PROBE_BED_POSITION 	30	
>BACK_PROBE_BED_POSITION 	180

>MESH_INSET			20	# Mesh inset margin on print area for MESH_BED_LEVELING

JanHH2019, 2019-03-11:
Thank you for your prompt reply. That's exactly how I configured it. When I go to Autolevel he drives to the HomePoint. However, the BLTouch is not activated, it does not drive the pen. However, I have put the BLTouch on the ZMax limit switch.

That's my configuration:

#=========================== Leveling button settings  
>cfg_leveling_mode			1			#1:auto-leveling; 0:manual leveling
#============================== Endstop  Settings 
>MIN_SOFTWARE_ENDSTOPS	 1              # 0:axes can move below MIN_POS;  1:axes won't move below MIN_POS.
>MAX_SOFTWARE_ENDSTOPS 	 1              # 0:axes can move below  MAX_POS;  1:axes won't move below MIN_POS.

# Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
>X_MIN_ENDSTOP_INVERTING 0		# set to true to invert the logic of the endstop.
>Y_MIN_ENDSTOP_INVERTING 1		# set to true to invert the logic of the endstop.
>Z_MIN_ENDSTOP_INVERTING 1		# set to true to invert the logic of the endstop.
>X_MAX_ENDSTOP_INVERTING 1		# set to true to invert the logic of the endstop.
>Y_MAX_ENDSTOP_INVERTING 1		# set to true to invert the logic of the endstop.
>Z_MAX_ENDSTOP_INVERTING 1		# set to true to invert the logic of the endstop.
>Z_MIN_PROBE_ENDSTOP_INVERTING 1	# set to true to invert the logic of the probe.
>FIL_RUNOUT_INVERTING 	0			# set to true to invert the logic of the Filament Runout Sensor.
# Specify here all the endstop connectors that are connected to any endstop or probe.
>USE_XMIN_PLUG	0			# 1:used; 0:noused
>USE_YMIN_PLUG	1			# 1:used; 0:noused
>USE_ZMIN_PLUG	1			# 1:used; 0:noused
>USE_XMAX_PLUG	1			# 1:used; 0:noused
>USE_YMAX_PLUG	0			# 1:used; 0:noused
>USE_ZMAX_PLUG	1			# 1:used; 0:noused

#============================= Z Probe Options =============================
>BLTOUCH			1			# 0:disable BLTOUCH; 1:enable BLTOUCH
#Select for a probe connected to Z-Min or Z-Max.
>Z_MIN_PROBE_PIN_MODE		2	# 0 : NULL; 1: ZMIN; 2: ZMAX

>Z_PROBE_OFFSET_FROM_EXTRUDER	0	# Z offset: -below +above  [the nozzle]		
>X_PROBE_OFFSET_FROM_EXTRUDER	-42.75	# X offset: -left  +right  [of the nozzle]
>Y_PROBE_OFFSET_FROM_EXTRUDER	0	# Y offset: -front +behind [the nozzle]
>XY_PROBE_SPEED 		4000	# X and Y axis travel speed (mm/m) between probes
>Z_PROBE_SPEED_FAST 		600	# Speed for the first approach when double-probing (with PROBE_DOUBLE_TOUCH)
>Z_PROBE_SPEED_SLOW 		300	# Speed for the "accurate" probe of each point

#=============================== Bed Leveling 
>BED_LEVELING_METHOD 		3	# 0:NULL_BED_LEVELING; 3:AUTO_BED_LEVELING_BILINEAR; 5:MESH_BED_LEVELING	

>GRID_MAX_POINTS_X 		3	# the number of grid points per dimension. <= 15
>GRID_MAX_POINTS_Y 		3	# the number of grid points per dimension. <= 15	
>Z_CLEARANCE_DEPLOY_PROBE	20	# Z Clearance for Deploy/Stow	   > 0
>Z_CLEARANCE_BETWEEN_PROBES	20	# Z Clearance between probe points  > 0

# Set the boundaries for probing (where the probe can reach).
>LEFT_PROBE_BED_POSITION 	50	
>RIGHT_PROBE_BED_POSITION 	160
>FRONT_PROBE_BED_POSITION 	230	
>BACK_PROBE_BED_POSITION 	180

>MESH_INSET			20	# Mesh inset margin on print area for MESH_BED_LEVELING

vk220, 2019-03-15:
Hi! Would you share please where you managed to buy Robin2? Thanks :-)

3Dangel, 2019-03-15:
Aliexpress

JanHH2019, 2019-03-16:
https://de.aliexpress.com/item/Motherboard-Robin2-STM32F407-32-bit-ARM-steuerung-bord-3-5-zoll-touch-screen-3D-drucker-teile/32969663990.html?spm=a2g0s.9042311.0.0.10cc4c4d12yKr8

vk220, 2019-03-16:
> 
> 
> https://de.aliexpress.com/item/Motherboard-Robin2-STM32F407-32-bit-ARM-steuerung-bord-3-5-zoll-touch-screen-3D-drucker-teile/32969663990.html?spm=a2g0s.9042311.0.0.10cc4c4d12yKr8

Thanks!

DaVincIan, 2019-03-31:
Jan H

Thank you for sharing your information, much appreciated

I might miss it, but did you solve your issue with "Homeing"?

I'm building a coreXY printer. Due to inexperience, I bought a MKS Robin. I since learned that only the Robin2 and not Robin can do BL touch (see post #75 for those who is still trying to figure this out).

Jan, please share the wire connection of the BL Touch to the Robin2.

Thanks
Ian

JanHH2019, 2019-04-06:
Hello DaVinclan,
in the picture yellow marked connection for bltouch.
n the case of the 3-pole connection, the yellow cable must be attached to the center of the board, the red to the center of the connection and the green to the outside. Depending on the configuration, the two-pole cable can be connected to the zmin or zmax limit switch

[image]

DaVincIan, 2019-04-06:
Thank you very much Jan H. 

I'm waiting for my Robin2 controller..It's in the mail. I will put it to the test when I receive it, probably only in 4 weeks time., A real pity MKS could not "solve" the issue with a firmware update for the Rodin controller.

Regards
Ian

DaVincIan, 2019-06-14:
Hi Jan

I'm making good progress with my CoreXY printer. The Robin2 seems to be working well. I have some difficulty setting and configuring the Z-height. 

My bigger problem seems to be the out of filament sensor. Without filament, the print will not start. This is not a problem. However, when the print runs out of filament, the print just continues. No action from the controller. This seems to have been reported by other users. Do you have a solution?

Ian

feste2000, 2019-10-17:
I have the same card, but even if I recognize the sensor does not work me automatic leveling. What am I wrong?
below my configurations. I have 2 Zmin (Zmin-Xmax) with switch, X and Y min and then touch sensor on Zmax.

#===================Language settings==========================================
>cfg_language_adjust_type		1	#multi-language(enable:1, disable:0)

>cfg_language_type			7	#languages setting,this configuration is valid when "cfg_multiple_language" is disabled.
						#(simplified Chinese:1; traditional Chinese:2; English:3; Russian:4; Spanish:5;French:6;Italian:7).

#===================function settings=======================================
>cfg_insert_det_module			0			#Assemble power outage module (1:mks 220det;0:mks pwc)

>cfg_have_ups_device			0			#enable UPS?(yes:1; no:0)

>cfg_print_over_auto_close		0			#auto-off when print finishes(1:yes£»0:no)

>cfg_filament_det0_trigger_level	0			#the level signal of  outage detection E0 module (1:high level;0:low level)
>cfg_filament_det1_trigger_level	0			#the level signal of  outage detection E1 module (1:high level;0:low level)

>cfg_mask_det_function			0			#the function of power detecting and filament detecting(1:disable;0:enable)

# the method of save the gcode data with power off(1:Capacitor storage; 0:File system save)
>cfg_pwroff_save_mode			0

>cfg_filament_load_length		100			#the lenght to extrude filament (mm),Max:2000mm
>cfg_filament_load_speed		800			#the speed to extrude filament(mm/min)
>cfg_filament_load_limit_temperature	200			#It is the minimum temperature to extrude filament .

>cfg_filament_unload_length		100			#It is the minimum temperature to extrude filament .
>cfg_filament_unload_speed		800			#the speed to retract filament(mm/min)
>cfg_filament_unload_limit_temperature	200			#It is the minimum temperature to retract filament .

#=========================== Leveling button settings  =============================

>cfg_leveling_mode			1			#1:auto-leveling; 0:manual leveling

#=========================== manual leveling ==================================

>cfg_point_number			5			#the point number of manual leveling(3,4,5 point available)

#the 5 point location of  manual leveling
>cfg_point1:10,10					
>cfg_point2:190,10
>cfg_point3:190,190
>cfg_point4:10,190
>cfg_point5:100,100

#========================== auto leveling ===================================

>cfg_auto_leveling_cmd:G28;G29;

#==================Display Effect(refer to MKS TFT datasheet)===================================
#user-defined function1
>setmenu_func1_display		1				#1:display this button£¬0:no

#each command must be separated by semicolon ";"
>setmenu_func1:M84;

>moreitem_pic_cnt			0				#the number of "more" button 

#edit command for 1~7 "More" button
#each command must be separated by semicolon ";"
#each icon command must be less than 200 bytes
>moreitem_button1_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button2_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button3_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button4_cmd:G28 X0;G28 Y0;G28 Z0;
[... 205 lines cut ...]
>BACK_PROBE_BED_POSITION 	180

>MESH_INSET			20	# Mesh inset margin on print area for MESH_BED_LEVELING

#============================== Delta Settings =============================
>DELTA_SEGMENTS_PER_SECOND	40      #--default
>DELTA_DIAGONAL_ROD		346.75	#Center-to-center distance of the holes in the diagonal push rods.	
>DELTA_SMOOTH_ROD_OFFSET	211.5	#Horizontal offset from middle of printer to smooth rod center.
>DELTA_EFFECTOR_OFFSET		28	#Horizontal offset of the universal joints on the end effector.
>DELTA_CARRIAGE_OFFSET		14.5	#Horizontal offset of the universal joints on the carriages.
>DELTA_RADIUS			169	#Horizontal distance bridged by diagonal push rods when effector is centered.	
>DELTA_HEIGHT			302	#height from z=0.00 to home position
>DELTA_PRINTABLE_RADIUS		125	#Print surface diameter/2 minus unreachable space (avoid collisions with vertical towers).
>DELTA_CALIBRATION_RADIUS	100	#set the radius for the calibration probe points - max 0.8 * DELTA_PRINTABLE_RADIUS 

#============================== Wifi Settings =============================
>CFG_WIFI_MODE			1			#wifi mode(0:sta;1:ap)
>CFG_WIFI_AP_NAME		TP-LINK_C944		#wifi name
>CFG_WIFI_KEY_CODE		makerbase		#wifi password

>CFG_CLOUD_ENABLE		1			#cloud service enable(0:disable;  1:enable)
>CFG_WIFI_CLOUD_HOST	        www.baizhongyun.cn	#cloud server url
>CFG_CLOUD_PORT			10086			#cloud server port

#end.