makerbase-mks/MKS-Robin #85: MKS Robin2

JanHH2019, 2019-03-16:
I have a problem with my MKS ROBIN2. When printing, it makes a very slow advance in some places. It looks like this happens when you switch from one printing site to another. I searched for the G-code, but the slowest feed is F1200. But he drives much slower, if I guess so maybe F200 or F400 if at all.
Maybe someone has experience with the MKS Robin2 and can say something? The G-Code I have created with SLIC3R and there at Speed the lowest set to 20mm / s.

[Test-50x50x2.txt](https://github.com/makerbase-mks/MKS-Robin/files/2974153/Test-50x50x2.txt)

JanHH2019, 2019-03-19:
Have the problem solved. The board is 32 bit processor but too slow for the drivers with 1/128 steps. In the many steps the CPU no longer calculates and it does not reach the specified feed rates. With 1/32 steps it runs without sneaking and much faster. Weak picture for the boar ....

makerbase-mks, 2019-03-21:
Have tried gcode with other slicing software
What model are you using the motor drive and how many subdivisions are set?
You can provide some information, we test here

DaVincIan, 2019-07-22:
Jan

Is your bed leveling working? I want to use MBL. I get it setup and do the probing. but when I'm printing, the controller does not compensate/correct Z axis.

Ian

JanHH2019, 2019-07-22:
Hello Ian, yes it works fine for me. I am still on vacation until the end of July and therefore I do not have the configuration at hand. When I'm back I can send it to you. Maybe that will help you.

3Dangel, 2019-07-31:
Hello folks,
I would be interested to receive the config file if possible.
I have the v2.1 with 1.0 TFT and just uploaded the 2.07 (Mks’) and it
stemmed not able to run correctly.
Having on board wifi connected to my router and getting an “error” on top
right corner that I don’t know what it is for.

Tested out Marlin2.0, got the SD card running. Would need to test as well.

Hope to heater back from you soon.
Thanks for your help!!
Robert

DaVincIan, 2019-08-05:
3Dangel

First confirm which MKS Robin you are using, The is a Robin with versions > V2.1. Then there is a Robin2. Robin2 is still at version 1. It is a controller that was developed to use a BLTouch probe.

Ian

DaVincIan, 2019-08-05:
Hi Jan
I don't think it is a problem with the config file, But I will try it. In the mean while I attached mine.
[robin2_cfg.TXT](https://github.com/makerbase-mks/MKS-Robin/files/3466605/robin2_cfg.TXT)

As I mentioned, everything is working fine. I can probe and retrieve the probe data. Bur during printing, the corrections are not applied. I did check, the bed leveling is active.

I wish we could get MKS development involved in the conversation.

Ian

3Dangel, 2019-08-05:
Hi Ian
I have the mks-Robin 2.1
Not the Robin2

The last MKS’ FW in the GitHub is 2.08

I have Also the FIRST generation of the Mks-Robin that is running on FW 1.02

I just need to know if anyone are still running on FW V2.0x from MKS
single or dual extruder, If you can provide some inside.

My present issues are:

On screen Preheating button + is not adding temp, as soon as you touch it,
it goes to 0.

Wifi :
Repetier host “connect”, but cannot get to G codes to cross over.
Ponterface cannot connect.

Serial com port : works for both  Repetier and Ponterface.
But they just access to the board directly. And no feedback info shown on
the touch screen.

I’m afraid that the board is not working properly for the MKs’ FW.
And before jumping onto Marlin’s, just wanted to rule out all
possibilities.

Thanks for your help
Bob

Thanks

DaVincIan, 2019-08-06:
Hi Bob

The Robin and Robin2 controllers are quite different w.r.t. their firmware. You will see if you look at the config files. Substantial differences.

I do have a Robin, but I'm not using it, so my experience is **very** limited, as a result I cannot assist you. This stream is relating more to Robin2 issues. Possibly not the ideal place to ask for Robin assistance.

Sorry man

Ian

3Dangel, 2019-08-06:
Hello Ian
Np!
I know the chance is slim.
Since the board went through many alterations in a very short time.

Mine had the wifi module installed on board and that alone would possibly
create a problem due to pins that might have moved and changed.
I have “all” the FW from 2.03,.04, .06, .07 & .08 and no luck.

Most likely will just use Marlin 2 on it.

Thanks for your help and time!!!
All the best!!
Bob

DaVincIan, 2019-08-06:
Hi Bob

I have my eye on the "new firmware" as well. Will you let me know if you
get it up and running. I mean get the compiler setup and config files
setup. I tried, but I don't seem to be able to download the files under the
"firmware: directory.

I hope to use just Arduino.

Maybe we can stay in touch directly: csveldman@gmail.com

Regards
Ian