makerbase-mks/MKS-Robin #86: Robin Nano and Bltouch

guenex, 2019-03-16:
Where can i find howto wire and setup bltouch or similar on Robin nano?

makerbase-mks, 2019-03-21:
Robin nano does not support bltouch

guenex, 2019-03-30:
Why this in cfg?
  #============================= Z Probe Options =============================
#BLTOUCH			0	# 0:disable BLTOUCH; 1:enable BLTOUCH

zillarob, 2019-04-12:
Pretty sure just to screw with people lol

matthewjlawr, 2019-05-29:
Yes, why is it in the config? There is a "1" to enable it. 
#============================= Z Probe Options =============================
#BLTOUCH			0	# 0:disable BLTOUCH; 1:enable BLTOUCH
#Select for a probe connected to Z-Min or Z-Max.
>Z_MIN_PROBE_PIN_MODE		2	# 0 : NULL; 1: ZMIN; 2: ZMAX

>Z_PROBE_OFFSET_FROM_EXTRUDER	0	# Z offset: -below +above  [the nozzle]		
>X_PROBE_OFFSET_FROM_EXTRUDER	0	# X offset: -left  +right  [of the nozzle]
>Y_PROBE_OFFSET_FROM_EXTRUDER	0	# Y offset: -front +behind [the nozzle]
>XY_PROBE_SPEED 		4000	# X and Y axis travel speed (mm/m) between probes
>Z_PROBE_SPEED_FAST 		600	# Speed for the first approach when double-probing (with PROBE_DOUBLE_TOUCH)
>Z_PROBE_SPEED_SLOW 		300	# Speed for the "accurate" probe of each point

zillarob, 2019-05-29:
I think somebody made some promises they couldnt keep. At best, ones they couldnt keep in a timely manner maybe =D

matthewjlawr, 2019-05-29:
Well if they can't check spelling in their released config files, there is little hope for the source code. Plus not sure why the file is a .txt extension instead of .cfg extension. But if BLTOUCH is not supported I don't think I'll upgrade to another one of their cards.

zillarob, 2019-05-29:
Ya, somebody needs to get their head checked for making the decision to keep it locked down.
Nice bit of gear, and the .txt cfg method is great for the new peeps, but it will never really go anywhere without the room to grow that being able to compile ourselves gives. 
Braindead move MKS =(

zillarob, 2019-08-10:
Got BLT working on Nano. 2 ways I have tested so far on the current main Marlin branch.
(Obviously doesnt work on the original fw, and I havent put any work into the dead end version here to see if it works, I would assume it does)

1- Easiest is to use the E1 Step pin for the servo if you arent using this for 2nd extruder or something.

2- A little tougher, but can also use Z- for the servo if you arent afraid of taking a soldering iron to your board. For some reason they used a very high value (10k) protection resistor on the S pin back to the chip. This seems to knock the signal down too much and causes the pin to be erratic. Knocking C35 and R45 off the board and jumpering the pads of R45 makes it happy. 
C35 aint no thing (Gen l has the pads for these, but they didnt put them on).
If you short R45 you will need to be careful afterward and not hook anything up to it incorrectly. Not a problem if you pay attention, but the lifeguard is no longer on duty without the resistor there to save you.

Pins file needs these added and set to suit whichever pin you choose to use. 
Normal BLT setup after that.
#define NUM_SERVOS                  ??  Probably 1
#define SERVO0_PIN                     ??  E1 STP = PA6, Z- = PA11
#define SERVO0_TIMER_NUM       ?? E1 STP = 8, Z- = 1

Gehring88, 2019-11-29:
@zillarob  can you show us with pictures where and how we install the bl touch at mks ronin nano for marlin 2.0 please. thanks

zillarob, 2019-11-30:
There are some extra pins next to e1 for an external driver. 
The one with "step" or "stp" next to it where you want the servo wire.
The rest is pretty standard, 2 go to an endstop of you choice, the other 2 are power and ground that I think I get from an extra endstop.

Gehring88, 2019-11-30:
@zillarob, is it possible to make a some instructions how to change the Firmware correctly. Thanks

zillarob, 2019-11-30:
Its pretty universal so pretty much the same as any other write up out there.
Iirc, commonly enable blt, what leveling type you want, z safe homing, probe points, margin from edge of the bed. and z offset (can do this one later on the controller).

The tougher part was figuring out that pin. That is what I pasted above from the pins file.
Might only need to change servo0 to pa6 here.

dnutt20x6, 2019-12-06:
@zillarob  I'm guessing this wouldn't be too much different for a Robin Mini?

I'm not clear on the timer number, or whether I need to remove any resistors, etc, like with the Nano.

zillarob, 2019-12-06:
Im not familiar with the mini, and quick peek at the schem shows it being quite a bit different.
Doesnt look like it has 5 drivers so PA6 at e1 is not an option. (this pin looks to be a vref for xy drivers on the mini)
I dont know if you can use one of the extra endstops as an alternate, but these are the ones you have to mod the board (remove/jumper resistors)

dnutt20x6, 2019-12-06:
@zillarob  OOF, I may wait and buy an ABL kit then, since they seem to work around a lot of these limitations without modding the board too much.  I can solder bigger things, but I'm still pretty bad at desoldering tiny SMD stuff.

I was using +Z as the servo pin and -Z s the endstop but can't get any response out of it, even after redirecting it.  It shows open and the pin redirects but seems to mess with the z stepper too much, prolly from the resistors issue you mentioned.

dnutt20x6, 2019-12-06:
Upon further thought, I figured screw it, I can't think of when I'll actually use the +Z for anything else, so no big loss.

I removed the R3 and R9 resistors from the +Z endstop and bridged the R9 pads and it worked.  My BLTouch clone responds to the menu commands, and stops the z-axis motor when homing!

Sweet.  Thanks for your help, @zillarob.

zillarob, 2019-12-06:
Glad to see you made that board your own.
Too many people are afraid to do this, or dont realize ow easy it is. 
Even if you do muck it up, not out much $, and rewarding to make it do what you need it to do, as opposed to what the mfg allows you to do.

dnutt20x6, 2019-12-07:
I'll add this as a final tip for anyone else who runs across this issue:

If your BLTouch/Clone is acting up, try tightening the little hex screw inside the middle hole on top.

Mine was leveling properly but throwing an error when I tried to print.  Tightening that screw fixed it.

If the probe can't keep it's pin up or struggles to do it, it'll throw an error.   That hex screw either closes the gap between the magnets inside or strengthens the induction effect that holds the pin up.  

Either way that's a possible fix.