makerbase-mks/MKS-Robin #88: Aren't you supposed to release the source if you use Marlin for your firmware?

zillarob, 2019-03-25:
Just poking around in there and see this. 
Not claiming to know what I am doing with a hex editor, but this looks really sketchy guys.

[image]

shaneelkaria, 2019-04-18:
Hey @zillarob can you help me with figuring out the M114 output for the TFT 28/32/35 ?
Can't get through with a solution. 
I'm using the RAMPS + TFT28 Config for now but cannot get through with the current X Y Z Positions when trying to use Pronterface through WiFi

zillarob, 2019-04-18:
That seems like it should just work if it were enabled in that version of the fw. Hopefully they didnt change things like that to make it work with their stuff.
I think Id try it over usb real quick to see if any different.
Figuring out why it doesnt work would probably be over my head unfortunately.