makerbase-mks/MKS-Robin #90: MKS Robin and BLTouch

uastalker, 2019-04-11:
When?

zillarob, 2019-04-12:
Wouldnt surprise me if they were working on their own version of the blt, but using a slightly different control sig to lock people into their ecosystem. Like Apple or something 3:)

It doesnt take a genius to know that if they were to release enough info for us to get an open source fw running on it, they would sell a ton more of them, and have a better brand rep.
I cant see much, if any, upside to locking people out of the opensource fw it is based on.
Whoever made that decision should prob go get their head checked =)

github051, 2021-02-09:
Finaly, I did it today. Bltouch is working with Mks Robin V1.4.

yexe21, 2023-02-17:
i have mks robin v2.4 (stm32f103zet6) control board. I want to connect BLTouch to this card, can you help me? Where should I connect the BLTouch servo cable?