makerbase-mks/MKS-Robin #95: Config and install fan on MKS ROBIN

andyrok, 2019-04-28:


andyrok, 2019-04-28:
I don’t  understand how to connect the fan....anyone help me?

DaVincIan, 2019-06-11:
Hi Andy

Look at the MKS Robin Data Sheet Pade 11: 
[MKS ROBIN DataSheet.pdf](https://github.com/makerbase-mks/MKS-Robin/files/3276065/MKS.ROBIN.DataSheet.pdf)

The fan that is controlled during printing and intended to blow on the print is wire to "Fan Port" (top left on data sheet P11). Make sure you have to polarity correct. It is indicated on the pc board. The positive is at the bottom (again, as per the orientation of the board in the aforementioned picture). 

Hope this helps
Ian

andyrok, 2019-06-11:
👍👍Many thanks

         Andrea R.
             
 
Si precisa che le informazioni di questa email sono riservate ad uso esclusivo del destinatario.
La comunicazione, la diffusione o la riproduzione in qualsiasi modo eseguite del messaggio, sono vietate. Eventuali violazioni saranno perseguite a norma di legge

elmo40, 2019-06-27:
Hello. Is it still the same M106 S### command? I tried multiple fans in both red/blk wire combinations and the fan never turns on.

DaVincIan, 2019-06-27:
MKS Robin & Robin2 uses Marlin GCODE: http://marlinfw.org/meta/gcode/

So yes M106 S### > 0 should let it run. The fan might not turn for small values. Begin with 255.
It uses a 12 V fan.
Can you verify that the fan in working, using a 12 V power supply?

elmo40, 2019-06-27:
I have 3 fully functional fans from other 3D printers. I have never seen it function on the Robin v2.4
Maybe the MOSFET was DOA?
My 'scope can see the bed and nozzle turning on but not a mV out of the fan terminals.

DaVincIan, 2019-06-27:
It's a bit of work,  but let me see if I can get my Robin wired up and test it. Not sure what the time line will be.Sent from my Samsung Galaxy smartphone.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/makerbase-mks/MKS-Robin/issues/95?email_source=notifications\u0026email_token=AGACKMUNRQIL2IWKA3LDSS3P4Q6NXA5CNFSM4HI53OJKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODYVUGGQ#issuecomment-506151706",
"url": "https://github.com/makerbase-mks/MKS-Robin/issues/95?email_source=notifications\u0026email_token=AGACKMUNRQIL2IWKA3LDSS3P4Q6NXA5CNFSM4HI53OJKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGODYVUGGQ#issuecomment-506151706",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

DaVincIan, 2019-07-01:
Hi Elmo40

I did a basic setup and it tested fine. I could control the fan from
Repetier Host Manual Control tab and sending GCode M106. M!07 turned it off.

Ian

elmo40, 2019-07-10:
I swapped the MOSFET. It was dead.
After a week of running with a fan, now the CPU is messed up.
It will run for a few layers then go all wonky. Time to toss it in the (ever growing) pile of low quality barely supported 32-bit boards...

DaVincIan, 2019-07-22:
Yip, so far I'm less impressed with mine as well. I had my MKS Robin 2 go coco on me. I reloaded the firmware. That worked for me. 

I can't get my Mesh leveling to work. I get the probing done. But the controller does not perform the leveling when printing.

Ian

3Dangel, 2019-07-22:
A lot of issues I find users are having.
Mine still works from the first purchase back couple years now.

I don’t do auto leveling ; use my hand and eye.
Loaded the first firmware and run since.

DaVincIan, 2019-07-25:
Thanks 3Dangel

For the most part my Robin2 is working. I have a 400 mm x 400 mm bed, so
bed leveling is essential. The controller does everything, performs the bed
leveling probing and saving the data. As far as I can determine, it is not
performing the "corrections" when printing. The status shows bed leveling
is active...but that's as far as it goes.

Ian

3Dangel, 2019-07-26:
I see my friend!
Have you tried if reducing the print bed size and try that way.
If it does work for smaller bed, it might be the calculation limitation in
the main board and related to the electronic’s capacity to resolve the
bigger area.
I used to have inductive’s and upgraded to photoelectric and back to
manual.
Of course, the promises is that you have to have “flat bed”. And for that,
I use glass.
Not printing ABS or other material, therefore, no much of bed temp as an
issue.
I still uses the first version of Mks-Robin  2016 when they first came out.
Version of FW was 2.x
The first stable version that brought the PIC for the temperature to
estabilize.

Hope you find out what is going on on yours.
Cheers

makerbase-mks, 2019-10-09:
The MKS Robin hardware schematic is open source, you can check it out: https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin