makerbase-mks/MKS-Robin #97: Strange Moving distances

Edge540, 2019-05-10:
Hello, just mount my MKS robin ( Firmware 2.0.8). Drivers are TMC2208. 
Microstep set to 1/16 (M1 High, M2 High, M3 Low). Stepper are Nema 17, 200 Stp/ with 20 teeth pulley . The step per mm is set to 80. 
When i move manually X and Y Axis with 10 mm request i have a move of 12,5 mm. Strange!!!!
If i change the step/mm in the .txt file, or remove the M1 M2 jumper no effect observed. The X and Y axis always move 12,5mm fort 10 mm requested.
If somebody have an idea , please help.

ThanKs

3Dangel, 2019-05-11:
> <https://maps.google.com/?q=16+(M1&entry=gmail&source=g> High, M2 High,

Did you change the config and then. Reload bin?

Edge540, 2019-05-11:
Yes many time Reload bin, changing Step/mm, testing with many configuration of jumper.!!

makerbase-mks, 2019-05-15:
If the distance you control does not correspond to the distance moved, it is wrong to set the number of steps, you can adjust the following

>XAXIS_STEPS_PER_MM		100				#X steps per mm 
>YAXIS_STEPS_PER_MM		100				#Y steps per mm 
>ZAXIS_STEPS_PER_MM		400				#Z steps per mm 
>EXT0_STEPS_PER_MM		100				#E0 steps per mm 
>EXT1_STEPS_PER_MM		100				#E1 steps per mm 

Set according to your actual situation

Edge540, 2019-05-24:
Thank's will try