makerbase-mks/MKS-Robin #99: MKS ROBIN2 knocks when printing circles

MisterMagomedov, 2019-05-24:
Good day! Installing and configuring the board did not cause any problems. However, when printing circles there is a very strong knock! This knock did not disappear, even when I changed different drivers, when I changed accelerations and jerks. The knock remained and is only present when the printing of circles begins. When printing on straight lines or diagonally, there are no knocks. The same situation is with my friend who bought Robin2.

kimer2002, 2019-05-25:
I have exactly the same problem on the printer after installing the micros Robin2 board. Strong knock when printing circles.
Tested on different drivers. Changed configuration, acceleration, jerk. Does not help.
I attach the video
https://www.youtube.com/watch?v=VuU9CssOD0Y

hobiseven, 2019-06-21:
Simple. I expext the same issue as in our alfawise porting. Is this an STM32F1? If yes the CPU skips an interrupt due special counter usage in the HAL. To fix it, it is likely the same as for Alfawise. See : 
#14030 in the issues. This is very likely it...