makerbase-mks/MKS-Robin #100: MKS Robin 2 comunication timout when printing over usb cable

vdc-fx, 2019-06-06:
Greetings.

On my Tronxy X5S. I was running standard controller with Octoprint.

After installing the MKS Robin 2.
Can print perfectly from the SD card.

From Octoprint and Pronterface. the print will stop with communication timeout.

On Octoprint, the print will continue after a couple of seconds. 
On Pronterface. the print will stop and does not continue.

The print seems to stop randomly at different points.

Is there a solution to this problem

Thanking you .
Vince

vdc-fx, 2019-06-15:
Reply from Makerbase

"
HI       
If the sd card prints   normally, the program on the motherboard should be no problem. If it is   printed on usb online, it will be caused by external interference, resulting   in abnormal data transmission, so it will stop printing.       
thanks
"
I have 3 MKS boards.

Old MKS, MKS GenL and MKS Robin2.

Use Raspberry PI and Windows 10 PC

Only the MKS Robin2 has this Fault.

Take it that the USB Ports has some interference during printing.

Will have to use this as a stand alone without PC connection.

github051, 2019-09-11:
For me also same issue with USB printing. SD card work ok. But while printing circles it makes click noise. Dont know what is the issue. 
I think Makerbase is without testing the boards they will sell.
All of their products support is very bad.

vdc-fx, 2020-01-25:
Tried Firmware dated 11/01/2019

MKS Robin is working fine now, printing with no problems.

But when I use the latest release. original problem, keeps losing communication.

Still using the old firmware . Hope they can resolve this problem.

I would use another controller, but then I cant use the screen from the Robin 2. (Flat ribbon cable)