makerbase-mks/MKS-Robin #104: Platformio fatal error with marlin 2 for MKS Robin

Edge540, 2019-06-24:
i have take configuration.h and configuration_adv.h here : https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin/firmware/Marlin2.0-MKS-Robin/Marlin
i usesource and tuto
platformio.ini with default "mks_robin"
Issue
Marlin\src\HAL\HAL_STM32F1\u8g_com_stm32duino_fsmc.cpp:35:10: fatal error: U8glib.h: No such file or directory

Any idea??

whhx602517961, 2019-06-26:
Is there a folder U8glib-HAL_ID1932 in the .piolibdeps folder?
Or you may download Marlin2.0-MKS-Robin folder and recompile the files.
I compiled the entire project without error.

Edge540, 2019-06-26:
Hi i confirm the folder U8glib-HAL_1932 is in the .piolibdeps ...
Always the same issue during compilation.

HailStorm32, 2019-07-04:
I had the same issue and was able to fix it by running `pio lib install "U8glib"`, but now I have a new issue

Edge540, 2019-07-04:
Hi Hailstorm32 thanks will try

Edge540, 2019-07-04:
Seems to be OK compilation OK Boot OK but now issues with the tourhcreen( RobinV203)..

mattdog01, 2019-12-30:
@Edge540 
Did you ever get Marlin 2.0 to work on your MKS-Robin v2.3 (I assume you have the v1.1 touchscreen)?
I keep getting the flashing blue light, by the reset button, on the main board and nothing other than "completed" on the touchscreen.

HailStorm32, 2019-12-30:
No, I gave up after hours of no luck. Ended up buying the mks sbase to replace the robin.

Edge540, 2020-01-04:
All is OK Now for me Working with good results

mattdog01, 2020-01-04:
@Edge540 
thanks for the note.  I now have my MKS Robin v2.3 and TFT v1.1 working perfectly and printing wonderfully with Marlin 2.0.  Huge thanks to jmz52 as that seems to be where most of the MKS-Robin Marlin development seems to have come from.  
I was able to get a huge amount of help from @jmz52 @sakza @tpruvot and @3Dangel under https://github.com/MarlinFirmware/Marlin/issues/16159  [BUG] ILI9328 TFT No longer functions MKS Robin #16159.  That issue has been resolved and merged to the branch.