makerbase-mks/MKS-Robin #107: Robin Mini cant compile

HailStorm32, 2019-07-04:
I'm getting the following errors when building.

`Marlin\src\HAL\HAL_STM32F1\u8g_com_stm32duino_fsmc.cpp:154:3: error: '__IO' does not name a type; did you mean '__tm'?`

`Marlin\src\HAL\HAL_STM32F1\u8g_com_stm32duino_fsmc.cpp:111:22: error: 'PD7' was not declared in this scope`

`Marlin\src\HAL\HAL_STM32F1\u8g_com_stm32duino_fsmc.cpp:136:22: error: 'PD11' was not declared in this scope`

`Marlin\src\HAL\HAL_STM32F1\u8g_com_stm32duino_fsmc.cpp:137:22: error: 'PD12' was not declared in this scope`

Any idea as to why, and how to fix it?

jfkerekes, 2019-08-05:
Hi,

I believe the Marlin 2.x bugfix is still missing several libraries that are needed to finish compiling firmware for the Robin Mini... I actually went back to the manufacturer of my printer after having the same issue compiling Marlin 2.x for my Robin Mini and the manufacturer sent me a copy of marlin with several extra libraries which allowed me to compile for the Robin Mini. It seems that the MKS Robin line of boards is "In the works" for Marlin but not finished from what I have read.

3Dangel, 2019-08-05:
Hi Jf
It would be interesting to post the working Marlin from your end and see if
Marlin Dev. Would take the portion that might be issuing and forge them
into the existing.

For sure Robin line are in the work. From everyone in the community like
yourself who brings in parts to merge it into the main.

Thanks for takin the time to report and support.

makerbase-mks, 2019-10-11:
You can use the MKS Robin Mini Marlin firmware from MKS Github or compile with Marlin's official website firmware.