makerbase-mks/MKS-Robin #108: Makerbase Robin 2 Marlin Problem

Onuryagci, 2019-07-11:
I need help. Mks robin 2 motherboard is not loading marlin firmware. Please help me. 
You certainly have knowledge. I'm good with apps.  Tell me what to do, please. Thank you.

Onuryagci, 2019-07-11:
Marlin and TMC 5160 Driver problem... Not moving motors. 
As if the drives were not plugged in.

DaVincIan, 2019-08-22:
Onuryagci

You are providing very little information. 
Are you using a Robin2 controller or a Robin V2 controller
I assume you managed to flash the controller with Marlin 2 FW. If so, to start with go through the Configuration.h file to make sure it the defines match your printer configuration.

3Dangel, 2019-08-23:
There is a section that you need to select what driver you are using.
Not sure if your driver is supported or not. Most likely yours is not that
popular one.
Cheers